microsoft/ELL #1: Doesn't build on a OSX Sierra Imac.

pquodling, 2017-06-30:
I already have llvm installed. and get the following. 

c-2%
Peters-iMac-2%
Peters-iMac-2% make
[  0%] Built target documentation
[ 11%] Built target utilities
[ 12%] Built target math
[ 15%] Built target data
[ 15%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRAssemblyWriter.cpp.o
In file included from /Users/peter/Projects/ELL/libraries/emitters/src/IRAssemblyWriter.cpp:17:
In file included from /Users/peter/Projects/ELL/libraries/emitters/include/IRModuleEmitter.h:15:
In file included from /Users/peter/Projects/ELL/libraries/emitters/include/IRFunctionEmitter.h:15:
/Users/peter/Projects/ELL/libraries/emitters/include/LLVMInclude.h:12:10: fatal error: 'llvm/Bitcode/ReaderWriter.h'
      file not found
#include "llvm/Bitcode/ReaderWriter.h"
         ^
1 error generated.
make[2]: *** [libraries/emitters/CMakeFiles/emitters.dir/src/IRAssemblyWriter.cpp.o] Error 1
make[1]: *** [libraries/emitters/CMakeFiles/emitters.dir/all] Error 2
make: *** [all] Error 2
Peters-iMac-2%

kmader, 2017-06-30:
I get further, but still crash at 24%

```
[ 24%] Building CXX object libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o
In file included from /Users/mader/Dropbox/Informatics/ELL/libraries/model/src/CompilableNode.cpp:9:
In file included from /Users/mader/Dropbox/Informatics/ELL/libraries/model/include/CompilableNode.h:11:
/Users/mader/Dropbox/Informatics/ELL/libraries/model/include/IRModelProfiler.h:15:10: fatal error: 
      'EmitterTypes.h' file not found
#include "EmitterTypes.h"
```

jacobrosenthal, 2017-06-30:
I can confirm this as well. I needed these changes:
https://github.com/Jacobrosenthal/ELL/tree/update-mac

And even then I had to remove the javascript bindings generation for a bunch of node/atom shell errors.

lovettchris, 2017-10-13:
Should be fixed now.

closed 2017-10-13 by lovettchris

HaRo87, 2017-12-11:
Hey folks,
I really like the idea of ELL. While trying to set it up I am struggling with getting it up and running on my Mac. I get the same error as already mentioned above:

`ELL/libraries/model/include/IRModelProfiler.h:15:10: fatal error: 
      'EmitterTypes.h' file not found
#include "EmitterTypes.h"`

The installation procedure on the Pi worked like a charm.

@lovettchris can you provide any details about how you've fixed that?

michhar, 2018-01-20:
@HaRo87 I'm on macOS High Sierra 10.13.2

The fix was pretty simple.  It has to do with where the llvm package gets installed.  To fix, I just symbolically linked from the llvm@3.9 (it names the Cellar this) into the llvm Cellar folder:

```
Fix:
cd /usr/local/Cellar/llvm
ln -s ../llvm@3.9/3.9.1_1
```
I was able then to run `cmake ..` and then `make` to build the latest ELL (v2.1.7) (checkout today).
Hop that helps.

@lovettchris May want to note in the macOS installation instructions.

kernhanda, 2018-01-20:
thanks @michhar. We'll add this to our notes and try to make this automatic for future builds.

YanMinge, 2018-01-23:
@michhar , I refer to your settings, but the same error still occurred.

HaRo87, 2018-01-28:
Hey @michhar and @YanMinge,
thanks for your support. I was able to build ELL now. But as already raised by @YanMinge on issue #130 I am now stuck at the point of compiling the my first model following this [tutorial](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/). :-(

luto65, 2018-05-22:
I confirm the issue of building on MacOS (10.13.3) and as well the resolution obtained by 
```
cd /usr/local/Cellar/
mkdir llvm
cd llvm
ln -s ../llvm@3.9/3.9.1_1
export PATH="/usr/local/Cellar/llvm/3.9.1_1/bin:$PATH"
cd <ELL distribution>/build
cmake ..
make
```