microsoft/ELL #2: Build failing in Ubuntu 16.04 32 Bit

vipings, 2017-06-30:
> Build failing on `[ 43%] Building CXX object libraries/nodes/CMakeFiles/nodes.dir/src/BinaryConvolutionalLayerNode.cpp.o`.  
Including the full stack trace [here](<script src="https://gist.github.com/vipings/4053446a334dbb6319b5707d6010b1d7.js"></script>)

DarrenRainey, 2017-07-01:
What version of gcc/g++ and cmake are you using sometimes old compilers won't work with new coding standardss

vipings, 2017-07-01:
> - `cmake version 3.5.1`  
   - `gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609`

DarrenRainey, 2017-07-01:
maybe try upgrading gcc to version 6 and cmake to 3.8 and check this guide that might help : https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md

lisaong, 2017-07-26:
Closing out this issue as it looks like an out of date cmake and g++.  @vipings, if you followed @DarrenRainey's recommendation, this should build for you.  If not, let us know.

closed 2017-07-26 by lisaong