microsoft/ELL #3: can't build on 64 bit ubuntu 16.04  - Any ideas..

ack72tdp, 2017-06-30:
root@ad00bef8edc1:/anil/ELL/build# gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@ad00bef8edc1:/anil/ELL/build# uname -a
Linux ad00bef8edc1 4.9.31-moby #1 SMP Sat Jun 24 06:29:23 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

root@ad00bef8edc1:/anil/ELL/build# cmake ..
-- The CXX compiler identification is unknown
-- Check for working CXX compiler: /usr/bin/gcc
-- Check for working CXX compiler: /usr/bin/gcc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCXXCompiler.cmake:54 (message):
  The C++ compiler "/usr/bin/gcc" is not able to compile a simple test
  program.

  It fails with the following output:

   Change Dir: /anil/ELL/build/CMakeFiles/CMakeTmp

  Run Build Command:"/usr/bin/make" "cmTC_2c297/fast"

  /usr/bin/make -f CMakeFiles/cmTC_2c297.dir/build.make
  CMakeFiles/cmTC_2c297.dir/build

  make[1]: Entering directory '/anil/ELL/build/CMakeFiles/CMakeTmp'

  Building CXX object CMakeFiles/cmTC_2c297.dir/testCXXCompiler.cxx.o

  /usr/bin/gcc -o CMakeFiles/cmTC_2c297.dir/testCXXCompiler.cxx.o -c
  /anil/ELL/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx

  gcc: error trying to exec 'cc1plus': execvp: No such file or directory

  CMakeFiles/cmTC_2c297.dir/build.make:65: recipe for target
  'CMakeFiles/cmTC_2c297.dir/testCXXCompiler.cxx.o' failed

  make[1]: *** [CMakeFiles/cmTC_2c297.dir/testCXXCompiler.cxx.o] Error 1

  make[1]: Leaving directory '/anil/ELL/build/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_2c297/fast' failed

  make: *** [cmTC_2c297/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/anil/ELL/build/CMakeFiles/CMakeOutput.log".
See also "/anil/ELL/build/CMakeFiles/CMakeError.log".
root@ad00bef8edc1:/anil/ELL/build# gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

clovett, 2017-07-02:
Looks like your GCC setup is broken, notice the cmake output:
> The C++ compiler "/usr/bin/gcc" is not able to compile a simple test program

This is what your cmake output should look like on Ubuntu 16.04:

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Blas libraries: /usr/lib/libblas.so
-- Blas linker flags:
-- Blas include directories:
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libblas.so
-- Using BLAS DLLs:
-- Found PythonInterp: /usr/bin/python (found version "2.7.12")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.12")
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12")
CMake Warning at interfaces/python/CMakeLists.txt:9 (message):
  Couldn't find Python libraries 3.5 or later, _ELL_python target will not be
  build properly.  This is not a required component, so it can be skipped.
-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /usr/bin/python
-- Using python libraries found at: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- Creating wrappers for python
CMake Warning at CMakeLists.txt:134 (message):
  Doxygen processor not found
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/d/git/ELL/ELL/linux
```

barronbarnett, 2017-07-02:
@ack72tdp 
Make sure G++ is installed.

It looks like you're using a docker container. I've created a [Dockerfile and repo](https://github.com/barronbarnett/ELL_Environment) that has all the proper tools already installed.

You can either reference it to fix you're current container or use it as a new starting point. I've also pushed it to docker hub so you can do `docker run -ti -v (yourlocalsourcedir):/root barronbarnett/ell_env:latest --name ell_env` and docker will pull down the image directly.

ack72tdp, 2017-07-05:
This is great Barron. Docker version worked perfect .. Thanks so much. Appreciate your help..

ack72tdp, 2017-07-06:
fixed using barron docker container.. that has all tools installed.

closed 2017-07-06 by ack72tdp