microsoft/ELL #4: make Does Not Work on Ubuntu Linux 14.04 x86_64

acs254, 2017-06-30:
I can build just fine into `> ELL/build`, but upon running `sudo make`, I get a huge list of errors -- too long to post here. But it ends with:

```
make[2]: *** [libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o] Error 1
make[1]: *** [libraries/utilities/CMakeFiles/utilities.dir/all] Error 2
make: *** [all] Error 2
```

devilloser, 2017-07-07:
so do i

lisaong, 2017-07-26:
Hi,

Thanks for bringing this up.

sudo make is not a supported build configuration for ELL at this time.  Is there a reason for needing to do `sudo make` instead of `make`?

Our make command does not install anything or require sudo privileges.

DarrenRainey, 2017-07-26:
can you share a pastebin with your error log

h4gen, 2017-07-28:
Hey! I have exactly the same issue but even without `sudo` I encounter the same problem. I tried to compile ELL in the CNTK docker container, provided by Microsoft, to build a consistent Toolchain in a transportable environment. I installed all dependencies in the [guide](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md). 
The container also uses Ubuntu 14.04.

lisaong, 2017-07-28:
Thanks for the details. Can you try a Ubuntu 16.04 container instead to see if it make a difference?

h4gen, 2017-07-29:
Hey @lisaong. With a fresh Ubuntu 16.04 everything works now. Thanks!

lisaong, 2017-07-29:
Great! Thanks for reporting back.

pnvphuong, 2017-07-30:
There are some incompatibilities with Ubuntu14.
ELL requires gcc 5 or above, but CUDA 8 only supports gcc 4x for Ubuntu 14. Therefore, it's hard to install and build ELL on Ubuntu 14.
On the other hand, CUDA 8 supports gcc 5x for Ubuntu 16. Everything will be fine then

lisaong, 2017-07-31:
Indeed, we are recommending Ubuntu 16.04 at minimum, unless there is a strong reason for using Ubuntu 14.

closed 2017-10-02 by lovettchris