microsoft/ELL #5: Error importing ELL (.\interfaces\python\test\test.py)

eulesv, 2017-06-30:
Hello,

I am able to build both ELL and interfaces on Windows, however, when I try to run tests on python it comes back with an error, is not able to load ELL
`Traceback (most recent call last):
  File "test.py", line 9, in <module>
    import ELL
  File "./..\ELL.py", line 17, in <module>
    _ELL = swig_import_helper()
  File "./..\ELL.py", line 16, in swig_import_helper
    return importlib.import_module('_ELL')
  File "C:\Users\ROBEP\AppData\Local\Continuum\Miniconda3\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: The specified module could not be found.`

The only reference I couldn't add is glibcc which is not available on the win-64 channel, any ideas on how to solve this?

Thanks!
--Rp

clovett, 2017-07-02:
Are you compiling ELL with Visual Studio 2015?  Since you are on windows you can run 'depend.exe' to see what dll is missing, run it on the _ELL.pyd module.  In case you don't have that handy tool, I attached it here:
[Depends.zip](https://github.com/Microsoft/ELL/files/1117505/Depends.zip)

eulesv, 2017-07-02:
Hi clovett, 

I'm using Visual Studio 2017, however I think the problem is related to missing packages on python, dependencies shows a bunch of missing modules:

- API-MS-WIN-CORE-APIQUERY-L1-1-0.DLL
- API-MS-WIN-CORE-APPCOMPAT-L1-1-1.DLL
- API-MS-WIN-CORE-APPINIT-L1-1-0.DLL
- API-MS-WIN-CORE-ATOMS-L1-1-0.DLL
- API-MS-WIN-CORE-COM-L1-1-0
- API-MS-WIN-CORE-COM-L1-1-1.DLL
- API-MS-WIN-CORE-COM-L1-1-2.DLL
- API-MS-WIN-CORE-COM-MIDLPROXYSTUB-L1-1-0.DLL
- API-MS-WIN-CORE-COM-PRIVATE-L1-1-0.DLL
- API-MS-WIN-CORE-COM-PRIVATE-L1-1-1.DLL
- API-MS-WIN-CORE-COMM-L1-1-0.DLL
- API-MS-WIN-CORE-CONSOLE-L2-1-0.DLL
- API-MS-WIN-CORE-CONSOLE-L3-1-0.DLL
- API-MS-WIN-CORE-CRT-L1-1-0.DLL
- API-MS-WIN-CORE-CRT-L2-1-0.DLL
- API-MS-WIN-CORE-DATETIME-L1-1-1.DLL
- API-MS-WIN-CORE-DATETIME-L1-1-2.DLL
- API-MS-WIN-CORE-DEBUG-L1-1-1.DLL
- API-MS-WIN-CORE-DELAYLOAD-L1-1-1.DLL
- API-MS-WIN-CORE-ENCLAVE-L1-1-0.DLL
- API-MS-WIN-CORE-ERRORHANDLING-L1-1-1.DLL
- API-MS-WIN-CORE-ERRORHANDLING-L1-1-3.DLL
- API-MS-WIN-CORE-FIBERS-L1-1-1.DLL
- API-MS-WIN-CORE-FIBERS-L2-1-1.DLL
- API-MS-WIN-CORE-FILE-L1-2-1.DLL
- API-MS-WIN-CORE-FILE-L1-2-2.DLL
- API-MS-WIN-CORE-FILE-L2-1-1.DLL
- API-MS-WIN-CORE-FILE-L2-1-2.DLL
- API-MS-WIN-CORE-HEAP-L1-2-0.DLL
- API-MS-WIN-CORE-HEAP-L2-1-0.DLL
- API-MS-WIN-CORE-HEAP-OBSOLETE-L1-1-0.DLL
- API-MS-WIN-CORE-INTERLOCKED-L1-2-0.DLL
- API-MS-WIN-CORE-IO-L1-1-1.DLL
- API-MS-WIN-CORE-JOB-L1-1-0.DLL
- API-MS-WIN-CORE-JOB-L2-1-0.DLL
- API-MS-WIN-CORE-KERNEL32-LEGACY-L1-1-1.DLL
- API-MS-WIN-CORE-KERNEL32-LEGACY-L1-1-5.DLL
- API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-1.DLL
- API-MS-WIN-CORE-KERNEL32-PRIVATE-L1-1-2.DLL
- API-MS-WIN-CORE-LARGEINTEGER-L1-1-0.DLL
- API-MS-WIN-CORE-LIBRARYLOADER-L1-2-0.DLL
- API-MS-WIN-CORE-LIBRARYLOADER-L1-2-2.DLL
- API-MS-WIN-CORE-LIBRARYLOADER-L2-1-0.DLL
- API-MS-WIN-CORE-LOCALIZATION-L1-2-1.DLL
- API-MS-WIN-CORE-LOCALIZATION-L1-2-2.DLL
- API-MS-WIN-CORE-LOCALIZATION-L2-1-0.DLL
- API-MS-WIN-CORE-LOCALIZATION-OBSOLETE-L1-3-0.DLL
- API-MS-WIN-CORE-LOCALIZATION-PRIVATE-L1-1-0.DLL
- API-MS-WIN-CORE-MEMORY-L1-1-2.DLL
- API-MS-WIN-CORE-MISC-L1-1-0.DLL
- API-MS-WIN-CORE-NAMEDPIPE-L1-2-0.DLL
- API-MS-WIN-CORE-NAMEDPIPE-L1-2-2.DLL
- API-MS-WIN-CORE-NAMESPACE-L1-1-0.DLL
- API-MS-WIN-CORE-NORMALIZATION-L1-1-0.DLL
- API-MS-WIN-CORE-PATH-L1-1-0.DLL
- API-MS-WIN-CORE-PERFCOUNTERS-L1-1-0.DLL
- API-MS-WIN-CORE-PRIVATEPROFILE-L1-1-1.DLL
- API-MS-WIN-CORE-PROCESSENVIRONMENT-L1-2-0.DLL
- API-MS-WIN-CORE-PROCESSSNAPSHOT-L1-1-0.DLL
- API-MS-WIN-CORE-PROCESSTHREADS-L1-1-2.DLL
- API-MS-WIN-CORE-PROCESSTHREADS-L1-1-3.DLL
- API-MS-WIN-CORE-PROCESSTOPOLOGY-L1-2-0.DLL
- API-MS-WIN-CORE-PSAPI-ANSI-L1-1-0.DLL
- API-MS-WIN-CORE-PSAPI-L1-1-0.DLL
- API-MS-WIN-CORE-PSM-KEY-L1-1-0.DLL
- API-MS-WIN-CORE-QUIRKS-L1-1-0.DLL
[... 333 lines cut ...]
- EXT-MS-WIN-SETUPAPI-INF-L1-1-0.DLL
- EXT-MS-WIN-SETUPAPI-INF-L1-1-1.DLL
- EXT-MS-WIN-SHELL-DIRECTORY-L1-1-0.DLL
- EXT-MS-WIN-SHELL-EMBEDDEDMODE-L1-1-0.DLL
- EXT-MS-WIN-SHELL-SHELL32-L1-2-0.DLL
- EXT-MS-WIN-SHELL-SHLWAPI-L1-2-0.DLL
- EXT-MS-WIN-SHELL32-SHELLCOM-L1-1-0.DLL
- EXT-MS-WIN-SHELL32-SHELLFOLDERS-L1-2-0.DLL
- EXT-MS-WIN-SMBSHARE-BROWSERCLIENT-L1-1-0.DLL
- EXT-MS-WIN-STORAGE-SENSE-L1-2-0.DLL
- EXT-MS-WIN-SXS-OLEAUTOMATION-L1-1-0.DLL
- EXT-MS-WIN-TSF-MSCTF-L1-1-2.DLL
- EXT-MS-WIN-UI-VIEWMANAGEMENT-L1-1-0.DLL
- EXT-MS-WIN-USP10-L1-1-0.DLL
- EXT-MS-WIN-WER-UI-L1-1-0.DLL
- EXT-MS-WIN-WER-XBOX-L1-1-0.DLL
- EXT-MS-WIN-WINRT-STORAGE-L1-1-0.DLL
- EXT-MS-WIN-WLAN-ONEXUI-L1-1-0.DLL
- EXT-MS-WIN-WLAN-SCARD-L1-1-0.DLL
- IESHIMS.DLL

So probably/hopefully one or more packages should be the answer to my problems :) any ideas?

Thanks for your help!
--Rp

clovett, 2017-07-02:
Ah, those are windows redirect dlls you can ignore those, but did any show up with yellow icons?  Let's also check that you actually built ELL.py and _ELL.pyd by running the following the additional steps listed in the [interfaces readme](https://github.com/Microsoft/ELL/blob/master/interfaces/README.md).

clovett, 2017-07-02:
For example, if you see this:
[image]

Then you should get this output when you try and run the demo:

```
(py35) D:\git\ELL\ELL\build\tutorials\vision\gettingStarted>python darknetDemo.py
====================================================================
ELL module is not loading
It is possible that you need to add LibOpenBLAS to your system path (See Install-*.md) from root of this repo
We have Intel versions here: D:\git\ELL\ELL\external\OpenBLASLibs.0.2.19.2\build\native\x64
You need to use the one that matches your computer chip model
   'haswell' works for haswell and broadwell chips
   'sandybridge' works for sandy bridge and ivy bridge chips
====================================================================
```
If this is the case then you need to do this:
set PATH=%PATH%;D:\git\ELL\external\OpenBLASLibs.0.2.19.2\build\native\x64\haswell\bin

(change the path to point to your ELL git repo location, and switch to "sandybridge\bin" if the above doesn't work).

eulesv, 2017-07-02:
Well, 
LIBOPENBLAS was missing but I tried setting it on the script using a 
`sys.path.append('./../../../bin/Debug')`
I know very little python so I don't know if it has the same effect, I'm going to try adding it to path and I'll let you know, thanks for your time clovett!
--Rp

eulesv, 2017-07-02:
There you go!! so sys.path.append only works for python modules? Anyhow, thank you very much clovett!! Now I can continue with the tutorial!

closed 2017-07-02 by eulesv