microsoft/ELL #7: error on ubuntu 16.04 x64

jackcc, 2017-07-01:
hi,i build ell on ubuntu  x64 , i fllow the install tutorial,i can start build ,but one error arise when it compile 64%:
[ 63%] Building CXX object libraries/common/CMakeFiles/common_test.dir/test/src/LoadDataset_test.cpp.o
[ 63%] Building CXX object libraries/common/CMakeFiles/common_test.dir/test/src/LoadMap_test.cpp.o
[ 64%] Building CXX object libraries/common/CMakeFiles/common_test.dir/test/src/LoadModel_test.cpp.o
[ 64%] Linking CXX executable common_test
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Copy(int, float const*, int, float*, int)’中：
BlasWrapper.cpp:(.text+0x31)：对‘cblas_scopy’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Copy(int, double const*, int, double*, int)’中：
BlasWrapper.cpp:(.text+0x69)：对‘cblas_dcopy’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Asum(int, float const*, int)’中：
BlasWrapper.cpp:(.text+0x92)：对‘cblas_sasum’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Asum(int, double const*, int)’中：
BlasWrapper.cpp:(.text+0xba)：对‘cblas_dasum’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Nrm2(int, float const*, int)’中：
BlasWrapper.cpp:(.text+0xf0)：对‘cblas_snrm2’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Nrm2(int, double const*, int)’中：
BlasWrapper.cpp:(.text+0x118)：对‘cblas_dnrm2’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Scal(int, float, float*, int)’中：
BlasWrapper.cpp:(.text+0x15b)：对‘cblas_sscal’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Scal(int, double, double*, int)’中：
BlasWrapper.cpp:(.text+0x193)：对‘cblas_dscal’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Axpy(int, float, float const*, int, float*, int)’中：
BlasWrapper.cpp:(.text+0x1d9)：对‘cblas_saxpy’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Axpy(int, double, double const*, int, double*, int)’中：
BlasWrapper.cpp:(.text+0x221)：对‘cblas_daxpy’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Dot(int, float const*, int, float const*, int)’中：
BlasWrapper.cpp:(.text+0x259)：对‘cblas_sdot’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Dot(int, double const*, int, double const*, int)’中：
BlasWrapper.cpp:(.text+0x290)：对‘cblas_ddot’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Gemv(CBLAS_LAYOUT, CBLAS_TRANSPOSE, int, int, float, float const*, int, float const*, int, float, float*, int)’中：
BlasWrapper.cpp:(.text+0x309)：对‘cblas_sgemv’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Gemv(CBLAS_LAYOUT, CBLAS_TRANSPOSE, int, int, double, double const*, int, double const*, int, double, double*, int)’中：
BlasWrapper.cpp:(.text+0x37c)：对‘cblas_dgemv’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Gemm(CBLAS_LAYOUT, CBLAS_TRANSPOSE, CBLAS_TRANSPOSE, int, int, int, float, float const*, int, float const*, int, float, float*, int)’中：
BlasWrapper.cpp:(.text+0x3f5)：对‘cblas_sgemm’未定义的引用
../math/libmath.a(BlasWrapper.cpp.o)：在函数‘ell::math::Blas::Gemm(CBLAS_LAYOUT, CBLAS_TRANSPOSE, CBLAS_TRANSPOSE, int, int, int, double, double const*, int, double const*, int, double, double*, int)’中：
BlasWrapper.cpp:(.text+0x471)：对‘cblas_dgemm’未定义的引用
collect2: error: ld returned 1 exit status
libraries/common/CMakeFiles/common_test.dir/build.make:190: recipe for target 'libraries/common/common_test' failed
make[2]: *** [libraries/common/common_test] Error 1
CMakeFiles/Makefile2:264: recipe for target 'libraries/common/CMakeFiles/common_test.dir/all' failed
make[1]: *** [libraries/common/CMakeFiles/common_test.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

I have already installed openblas。

cjacobs, 2017-07-01:
Hi,

Could you `cd` to your build directory, run `cmake ..` and paste the result here?

Thanks!

jackcc, 2017-07-01:
Thank you for your reply, this is the log

-- Blas libraries: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
-- Blas linker flags: 
-- Blas include directories: /usr/include
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
-- Using BLAS DLLs: 
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
CMake Warning at CMakeLists.txt:83 (message):
  SWIG not found, interfaces and projects that depend on them will not build
  properly.  They are not part of the default make targets, so can be
  skipped.  If you want to build interfaces (such as the Python bindings),
  follow the instructions in INSTALL-XXXXX.md file to install SWIG.  Then
  delete the CMakeCache.txt file from your build directory run the cmake
  command again.

-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
CMake Warning at interfaces/python/CMakeLists.txt:9 (message):
  Couldn't find Python libraries 3.5 or later, _ELL_python target will not be
  build properly.  This is not a required component, so it can be skipped.

-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for python
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for javascript
-- Found npm at /usr/bin/npm
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for xml
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jack/ELL/build

vipings, 2017-07-01:
You need to install SWIG.  
- Download SWIG from [here](http://www.swig.org/download.html)  
-  `tar zxvf swig-3.0.12.tar.gz && cd swig-3.0.12.tar`
- `./configure --without-pcre && make && sudo make install`

jackcc, 2017-07-01:
from the log: 
SWIG:This is not a required component, so it can be skipped.

SWIG must be installed?i will try it.

jackcc, 2017-07-01:
Thanks,@vipings,it works.but the interface is python , which i want is C++. any more suggestions?

vipings, 2017-07-01:
AFAIK ELL supports  JS, Python. This project is WIP and they would add more bindings. This is my assumption, by looking into the interface directory of ELL and I would recommend you to reach out ELL guys to get more accurate answer on this. Thanks. 😄

lisaong, 2017-07-02:
My guess is that the blas headers being picked up are incorrect. They aren't declaring functions that ELL relies on. 

BlasWrapper.cpp:(.text+0x31)：对‘cblas_scopy’未定义的引用

If you followed the INSTALL.*md and installed blas, maybe you had another version that is taking precedence (cmake found that instead of the one ELL needs). Try:

1) grep for that function in the header files to find the header needed, and
2a) if found, modify the Blas cmake in ELL/CMake to point to the corrected include and and library paths, or
2b) if not found, try installing blas again. If you search the issues thread for other recent Ubuntu build issues, you should find an example cmake output that someone posted which has the right blas paths.

jackcc, 2017-07-03:
@lisaong you are right，thank you very much.Before I installed openblas, I have installed atlas.so i remove atlas,it works very well.

closed 2017-07-03 by jackcc