microsoft/ELL #11: cmake fails on Ubuntu 17.04

malynda, 2017-07-02:
After tracking down SWIG and the ./configure --without-pcre , I'm still unable to successfully complete cmake. 

 cmake ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libopenblas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libopenblas.so
-- Using BLAS DLLs: 
CMake Error at CMake/LLVMSetup.cmake:41 (string):
  string sub-command REPLACE requires at least four arguments.
Call Stack (most recent call first):
  CMake/LLVMSetup.cmake:51 (get_llvm_config_list)
  CMakeLists.txt:63 (include)

lisaong, 2017-07-02:
Thanks for pasting the config details, this looks like a cmake error.  My guess is you have a newer version that is unhappy with line 51 of LLVMSetup.cmake. We've been using cmake 3.9 mostly, what version do you have and can you try 3.9?

malynda, 2017-07-02:
I've been running with cmake v3.7.2. Downloading the cmake rc5 for 3.9.0 now...

malynda, 2017-07-02:
Woohoo, it worked!

closed 2017-07-02 by malynda