microsoft/ELL #14: make failed on Raspberry Pi 3B, Arch Linux ARM

codelast, 2017-07-02:
Hardware version: Raspberry Pi 3B
OS: Arch Linux ARM

I've installed the dependencies and the make process failed finally, and part of the command line output messages are as following, so does anyone knows how could I solve it? Thanks in advance.

> [ 29%] Linking CXX static library libfunctions.a
> [ 29%] Built target functions
> Scanning dependencies of target model
> [ 29%] Building CXX object libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o
> In file included from /root/resource/ELL/libraries/model/include/Model.h:13:0,
>                  from /root/resource/ELL/libraries/model/include/IRModelProfiler.h:11,
>                  from /root/resource/ELL/libraries/model/include/CompilableNode.h:11,
>                  from /root/resource/ELL/libraries/model/src/CompilableNode.cpp:9:
> /root/resource/ELL/libraries/model/include/PortElements.h: In member function ‘const ell::model::OutputPort<ValueType>* ell::model::PortElements<ValueType>::ReferencedPort() const’:
> /root/resource/ELL/libraries/model/include/PortElements.h:448:135: error: ‘ell::model::PortElementBase’ is not a base of ‘const ell::model::PortElements<ValueType>’
>  ncedPort() const { return static_cast<const OutputPort<ValueType>*>(this->PortElementBase::ReferencedPort()); }
>                                                                                             ^~~~~~~~~~~~~~
> make[2]: *** [libraries/model/CMakeFiles/model.dir/build.make:63: libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:875: libraries/model/CMakeFiles/model.dir/all] Error 2
> make: *** [Makefile:95: all] Error 2

lisaong, 2017-07-02:
Thanks for trying this out, and for putting in your time to get this to compile for Pi.  Instructions on running models for Raspberry Pi will be coming very shortly.

There are two parts to ELL.  1) The infrastructure for importing and compiling models (this runs on desktop environments, not the Pi) and 2) the models that get compiled to run on small devices (this runs on the Pi). 

Instructions for the 2nd part are starting to be collected at tutorials/vision/gettingStarted, but have not covered Pi yet. This part we are working on actively because it's about how to run the ML models. We're hoping to get something ready in a week or so. 

We'll try to clarify the docs as well. Basically you don't need a build ELL itself on Pi to get the models to run. The error you are seeing is likely due to needing an updated gcc.

codelast, 2017-07-02:
Actually my gcc version is quite new:

> gcc (GCC) 7.1.1 20170516

Since you said that I don't need to compile ELL on Pi, then OK, I know I did a meaningless thing, so just wait to see your updated doc.
Thank you.

lisaong, 2017-07-02:
Ah, looks like you may have helped uncover a gcc 7 issue in the meantime. It'll be helpful when we support gcc 7 for desktop. 

Thanks for spending the time on this!

closed 2017-07-11 by codelast