microsoft/ELL #15: ELL.py file not found

VedAustin, 2017-07-02:
When I run this command:
`python cntkDemo.py`

I get this error message:
`Could not find './../../../build/interfaces/python/ELL.py', did you follow the ELL Python Binding build instructions?`

However when I ran this earlier:
`ctest --build-config Debug`

100% of the tests passed. Can you let me know how I can obtain this `ELL.py` file?

clovett, 2017-07-02:
Ah yes, there is one more step for building Python wrappers, it is described in the [interfaces readme](https://github.com/Microsoft/ELL/blob/master/interfaces/README.md).

eulesv, 2017-07-02:
Hi VedAustin,

You still need to build the python interface, follow the instructions here [https://github.com/Microsoft/ELL/tree/master/interfaces](https://github.com/Microsoft/ELL/tree/master/interfaces). I'm still having some issues on that part, so please let me know if it works.

--Rp

VedAustin, 2017-07-02:
Yes .. Finally it is working!

Before I ran this: `cmake --build .`
I ran this: `cmake --build . --target _ELL_python --config Release`
The above command might have installed ELL.py

When I ran this: ` ctest --build-config Debug`
None of the test even ran. So I figured i will run this: ` cmake --build .` and reran this: ` ctest --build-config Debug`
All tests passed.

When I ran this step again : `python cntkDemo.py` I got a new error message about a openblas not being in the path, so I included it in the path. Now it is working as expected.

closed 2017-07-02 by VedAustin

YG-FLUSH, 2017-07-04:
In this [page](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/README.md)
, it contains a wrong link([ELL Python Language Bindings](https://github.com/Microsoft/ELL/blob/master/interfaces/README.MD)) which is 404。