microsoft/ELL #16: Instructions to build

VedAustin, 2017-07-02:
I have a clarification to make:
Looking at the instructions here:
https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md

After this line `NuGet will download the prerequisites into the ELL/external directory.`
I have to run:
`cmake -G "Visual Studio 15 2017 Win64" -DPROCESSOR_HINT=haswell ..`

Do I need to change my directory to `external` because if I stay in `ELL`, and run the above command then the file `CMakeLists.txt` file will not be found.

Also when building the ELL in the `build` folder:
`cmake -G "Visual Studio 14 2015 Win64" -DPROCESSOR_HINT=haswell ..`

However since I have Visual Studio 2017 installed, can we also use this instead:
`cmake -G "Visual Studio 15 2017 Win64" -DPROCESSOR_HINT=haswell ..`

clovett, 2017-07-02:
You are right we are missing these steps in the readme:
```
mkdir build
cd build
```
Then run cmake.  We do have the VS 2017 command line in the readme. But I'll get the above added.

eulesv, 2017-07-02:
I'll piggy back on this one... right at the end of the tutorial, it makes a reference to a build script (compile_darknetReference.cmd)... where is this script located?

VedAustin, 2017-07-02:
I found this under `build\tutorials\vision\gettingStarted`

But I also found a `tutorial` folder under the root `ELL`. So I am a bit confused about two `tutorials` folder being present.

eulesv, 2017-07-02:
The one at the root is the original folder, the one under build is a copy created during build; I'm following the contentes of a compiling.md file but the tutorial says there is a build script, i think this file is missing on the repo

eulesv, 2017-07-02:
Nevermind... I just realized those files where added after I originally clone the repo, I just did fetch/pull :)

closed 2017-07-15 by lovettchris