microsoft/ELL #17: make failed on Ubuntu 17.04

malynda, 2017-07-02:
After updating to CMake 3.9.0.rc5, cmake was able to run successfully. New error when trying make.

 make
Scanning dependencies of target documentation
[  0%] Built target documentation
Scanning dependencies of target utilities
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CommandLineParser.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CompressedIntegerList.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ConformingVector.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CStringParser.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Files.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Format.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IArchivable.cpp.o
[  5%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IndentedTextWriter.cpp.o
[  5%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerList.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerStack.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/JsonArchiver.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchive.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchiver.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/OutputStreamImpostor.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/PPMImageParser.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/RandomEngines.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Tokenizer.cpp.o
[ 10%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/TypeName.cpp.o
[ 10%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/UniqueId.cpp.o
[ 11%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Variant.cpp.o
[ 11%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/XmlArchiver.cpp.o
[ 13%] Linking CXX static library libutilities.a
[ 13%] Built target utilities
Scanning dependencies of target math
[ 13%] Building CXX object libraries/math/CMakeFiles/math.dir/src/BlasWrapper.cpp.o
[ 14%] Linking CXX static library libmath.a
[ 14%] Built target math
Scanning dependencies of target data
[ 15%] Building CXX object libraries/data/CMakeFiles/data.dir/src/Dataset.cpp.o
[ 15%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVector.cpp.o
[ 16%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVectorOperations.cpp.o
[ 16%] Building CXX object libraries/data/CMakeFiles/data.dir/src/GeneralizedSparseParsingIterator.cpp.o
[ 17%] Building CXX object libraries/data/CMakeFiles/data.dir/src/SequentialLineIterator.cpp.o
[ 17%] Building CXX object libraries/data/CMakeFiles/data.dir/src/TextLine.cpp.o
[ 18%] Building CXX object libraries/data/CMakeFiles/data.dir/src/WeightLabel.cpp.o
[ 18%] Linking CXX static library libdata.a
[ 18%] Built target data
Scanning dependencies of target evaluators
[ 18%] Building CXX object libraries/evaluators/CMakeFiles/evaluators.dir/src/AUCAggregator.cpp.o
[ 19%] Building CXX object libraries/evaluators/CMakeFiles/evaluators.dir/src/BinaryErrorAggregator.cpp.o
[ 19%] Linking CXX static library libevaluators.a
[ 19%] Built target evaluators
Scanning dependencies of target functions
[ 19%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/ElasticNetRegularizer.cpp.o
[ 20%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/HingeLoss.cpp.o
[ 20%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/L2Regularizer.cpp.o
[ 21%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/LogLoss.cpp.o
[ 21%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/SmoothHingeLoss.cpp.o
[ 22%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/SquaredLoss.cpp.o
[ 22%] Linking CXX static library libfunctions.a
[ 22%] Built target functions
Scanning dependencies of target model
[ 22%] Building CXX object libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o
In file included from /home/malynda/Development/ELL/libraries/model/include/CompilableNode.h:11:0,
                 from /home/malynda/Development/ELL/libraries/model/src/CompilableNode.cpp:9:
/home/malynda/Development/ELL/libraries/model/include/IRModelProfiler.h:15:26: fatal error: EmitterTypes.h: No such file or directory
 #include "EmitterTypes.h"
                          ^
compilation terminated.
libraries/model/CMakeFiles/model.dir/build.make:62: recipe for target 'libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o' failed
make[2]: *** [libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o] Error 1
CMakeFiles/Makefile2:739: recipe for target 'libraries/model/CMakeFiles/model.dir/all' failed
make[1]: *** [libraries/model/CMakeFiles/model.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

clovett, 2017-07-02:
Weird, I'm not getting this error.  But my build built things in a different order, it did "emitters" after "data", so I wonder if that's why it found the header.  The file lives here:
`./libraries/emitters/include/EmitterTypes.h`

The ./libraries/model/CMakeLists.txt contains:
`target_include_directories(${library_name} PUBLIC include)`

So I wonder if adding this include path will help make it less dependent on build order:
`target_include_directories(${library_name} PUBLIC include ../emitters/include)`

malynda, 2017-07-02:
@clovett what version of Ubuntu are you using? I'm now trying this on my 16.04 box and everything is going swimmingly so far.

malynda, 2017-07-04:
I discovered I didn't have llvm installed... went back, installed, re-did cmake and was able to run make just fine.

closed 2017-07-04 by malynda