microsoft/ELL #21: No module named 'darknet_to_ell'

malynda, 2017-07-02:
I just built everything and was going to test the darknet tutorial, but alas! errors.

$ python darknetDemo.py 
Traceback (most recent call last):
  File "darknetDemo.py", line 7, in <module>
    import darknet_to_ell
ModuleNotFoundError: No module named 'darknet_to_ell'

My guess is it doesn't know that darknet_to_ell is in /build/tools/importers/darknet
Not sure if this was supposed to be copied in the python build or if it needs to be directed to the correct folder.

eulesv, 2017-07-02:
Did you use the rebuild.cmd? It uses a different configuration RelWithDebInfo, none of the scripts point to that one :(

malynda, 2017-07-02:
Nope, just followed what the Python Language Bindings instructions on https://github.com/Microsoft/ELL/blob/master/interfaces/README.md said.

jimaobian, 2017-07-03:
@malynda There is a typo in findEll.py :
`sys.path.append(buildDir + '/tools/importers/Darknet')`

Change `Darknet` to `darknet`

malynda, 2017-07-03:
Thanks @jimaobian !  Have you already submitted a PR for that?

closed 2017-07-03 by malynda