microsoft/ELL #27: Fail to run sh compile_darknetReference.sh on Mac OSX

jimaobian, 2017-07-03:
* `python darknetDemo.py` works well.
* `sh compile_darknetReference.sh` fails to run on the last step, namely:
`/Users/xxx/anaconda/bin/python darknetReference_setup.py build install
running build`
* Here is the error details:
```
/Users/XXX/ELL/interfaces/common/CallbackInterface.h:28:35: warning: defaulted function
      definitions are a C++11 extension [-Wc++11-extensions]
        virtual ~CallbackBase() = default;
                                  ^
/Users/XXX/ELL/interfaces/common/CallbackInterface.h:55:40: warning: defaulted function
      definitions are a C++11 extension [-Wc++11-extensions]
        virtual ~CallbackForwarder() = default;
                                       ^
In file included from darknetReference_wrap.cpp:6144:
./darknetReference_wrap.h:22:13: error: exception specification of overriding function is more lax than base version
    virtual ~SwigDirector_DoubleCallbackBase();
            ^
/Users/XXX/ELL/interfaces/common/CallbackInterface.h:28:17: note: overridden virtual function is
      here
        virtual ~CallbackBase() = default;
                ^
In file included from darknetReference_wrap.cpp:6144:
./darknetReference_wrap.h:64:13: error: exception specification of overriding function is more lax than base version
    virtual ~SwigDirector_FloatCallbackBase();
            ^
/Users/XXX/ELL/interfaces/common/CallbackInterface.h:28:17: note: overridden virtual function is
      here
        virtual ~CallbackBase() = default;
                ^
```

jacobrosenthal, 2017-07-06:
Sierra 10.12.5
Im stuck exactly here as well

Though Ill note, I had one more snag even getting here
```
compile_darknetReference.sh: line 9: llc: command not found
```
Apparently on mac with brew, llvm bins are not linked https://github.com/Homebrew/legacy-homebrew/issues/29733

I had to change line 9 from
```
llc -filetype=obj darknetReference.ll
```
to 
```
/usr/local/opt/llvm\@3.9/bin/llc -filetype=obj darknetReference.ll
```

jacobrosenthal, 2017-07-10:
The instructions no longer mention ```sh compile_darknetReference.sh``` anymore, and whats in the readme works for me now.

lisaong, 2017-07-26:
Thanks for letting us know! Closing out this issue as resolved.  The compile_darknetReference.sh file was temporary and we've replaced it with the (hopefully more robust) cmake mechanism.

closed 2017-07-26 by lisaong