microsoft/ELL #29: Error while compiling

sria91, 2017-07-03:
I'm getting the following error while executing `make _ELL_python` as per the instructions available at https://github.com/Microsoft/ELL/blob/master/interfaces/README.md

```sh
[ 20%] Built target utilities
[ 22%] Built target math
[ 29%] Built target data
[ 31%] Built target evaluators
[ 37%] Built target functions
[ 39%] Building CXX object libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o
In file included from /home/sria91/Development/ELL/libraries/model/include/CompilableNode.h:11:0,
                 from /home/sria91/Development/ELL/libraries/model/src/CompilableNode.cpp:9:
/home/sria91/Development/ELL/libraries/model/include/IRModelProfiler.h:15:26: fatal error: EmitterTypes.h: No such file or directory
compilation terminated.
libraries/model/CMakeFiles/model.dir/build.make:62: recipe for target 'libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o' failed
make[3]: *** [libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o] Error 1
CMakeFiles/Makefile2:707: recipe for target 'libraries/model/CMakeFiles/model.dir/all' failed
make[2]: *** [libraries/model/CMakeFiles/model.dir/all] Error 2
CMakeFiles/Makefile2:2203: recipe for target 'interfaces/python/CMakeFiles/_ELL_python.dir/rule' failed
make[1]: *** [interfaces/python/CMakeFiles/_ELL_python.dir/rule] Error 2
Makefile:584: recipe for target '_ELL_python' failed
make: *** [_ELL_python] Error 2
```

Kindly help.

malynda, 2017-07-03:
Hi! I had this issue also, on Ubuntu 17.04.  What OS & version are you running?

sria91, 2017-07-04:
@malynda I'm running Ubuntu 16.04

lovettchris, 2017-07-06:
This can happen if you have missing prerequisite libraries (openblas, LLVM, etc).  Perhaps openblas is not optional as per the [install instructions](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md).

sria91, 2017-07-06:
@lovettchris  I've installed openBLAS as well as doxygen. Still I am getting the error.

lovettchris, 2017-07-06:
And you got LLVM ?

```
sudo apt-get install -y wget
wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key | sudo apt-key add -
sudo apt-add-repository "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-3.9 main"
sudo apt-get -y update
sudo apt-get install -y llvm-3.9-dev
```

sria91, 2017-07-06:
@lovettchris Yes, I do.

jacobmanning, 2017-07-06:
What is the output of your `cmake ..` command? Has CMake correctly found your Python libraries?

sria91, 2017-07-11:
@jacobmanning Following is the output of the cmake command:
```sh
sria91@gpumachine:~/Development/ELL/build$ cmake -D PYTHON_EXECUTABLE=/usr/bin/python3 ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
-- Using BLAS DLLs: 
-- Found PythonInterp: /usr/bin/python3 (found version "3.5.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found version "3.5.2") 
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /usr/bin/python3
-- Using python libraries found at: /usr/lib/x86_64-linux-gnu/libpython3.5m.so
-- Creating wrappers for python
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sria91/Development/ELL/build
```
Of course it finds the Python libraries.

lovettchris, 2017-07-15:
Do you still have this problem on the latest bits?

sria91, 2017-07-17:
@lovettchris yes. Following is the output of the `make` command on latest bits.
```sh
sria91@gpumachine:~/Development/ELL/build$ make -j4
Scanning dependencies of target documentation
Scanning dependencies of target utilities
Scanning dependencies of target modelFiles
Scanning dependencies of target testing
[  0%] Built target documentation
[  0%] Building CXX object libraries/testing/CMakeFiles/testing.dir/src/testing.cpp.o
[  0%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CommandLineParser.cpp.o
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CompressedIntegerList.cpp.o
[  1%] Built target modelFiles
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ConformingVector.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CStringParser.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Files.cpp.o
[  2%] Linking CXX static library libtesting.a
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Format.cpp.o
[  2%] Built target testing
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IArchivable.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IndentedTextWriter.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerList.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerStack.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/JsonArchiver.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchive.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchiver.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/OutputStreamImpostor.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/PPMImageParser.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/RandomEngines.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Tokenizer.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/TypeName.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/UniqueId.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Variant.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/XmlArchiver.cpp.o
[ 10%] Linking CXX static library libutilities.a
[ 10%] Built target utilities
Scanning dependencies of target math
Scanning dependencies of target emitters
[ 10%] Building CXX object libraries/math/CMakeFiles/math.dir/src/BlasWrapper.cpp.o
[ 12%] Linking CXX static library libmath.a
[ 12%] Built target math
Scanning dependencies of target math_test
Scanning dependencies of target data
Scanning dependencies of target functions
[ 12%] Building CXX object libraries/math/CMakeFiles/math_test.dir/test/src/main.cpp.o
[ 13%] Building CXX object libraries/data/CMakeFiles/data.dir/src/Dataset.cpp.o
[ 13%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/ElasticNetRegularizer.cpp.o
[ 13%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/HingeLoss.cpp.o
[ 13%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVector.cpp.o
[ 14%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/L2Regularizer.cpp.o
[ 15%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/EmitterTypes.cpp.o
[ 15%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRAssemblyWriter.cpp.o
[ 16%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVectorOperations.cpp.o
[ 16%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/LogLoss.cpp.o
[ 18%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/SmoothHingeLoss.cpp.o
[ 18%] Building CXX object libraries/functions/CMakeFiles/functions.dir/src/SquaredLoss.cpp.o
[ 18%] Linking CXX static library libfunctions.a
[ 18%] Built target functions
Scanning dependencies of target functions_test
[ 19%] Building CXX object libraries/functions/CMakeFiles/functions_test.dir/test/src/main.cpp.o
[ 19%] Building CXX object libraries/data/CMakeFiles/data.dir/src/GeneralizedSparseParsingIterator.cpp.o
[ 19%] Building CXX object libraries/data/CMakeFiles/data.dir/src/SequentialLineIterator.cpp.o
[ 19%] Linking CXX executable math_test
[ 20%] Building CXX object libraries/data/CMakeFiles/data.dir/src/TextLine.cpp.o
[ 20%] Linking CXX executable functions_test
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, float const*, int, float*, int)':
BlasWrapper.cpp:(.text+0x31): undefined reference to `cblas_scopy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, double const*, int, double*, int)':
BlasWrapper.cpp:(.text+0x69): undefined reference to `cblas_dcopy'
../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Asum(int, float const*, int)':
BlasWrapper.cpp:(.text+0x92): undefined reference to `cblas_sasum'
[... 69 lines cut ...]
[ 20%] Building CXX object libraries/data/CMakeFiles/data.dir/src/WeightLabel.cpp.o
[ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRBlockRegion.cpp.o
[ 21%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRDiagnosticHandler.cpp.o
[... cut ...]
[ 27%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRSwigInterfaceWriter.cpp.o
[ 28%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/ModuleEmitter.cpp.o
[ 28%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/TargetDevice.cpp.o
[ 30%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/Variable.cpp.o
[ 30%] Linking CXX static library libemitters.a
[ 30%] Built target emitters
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

hustzxd, 2017-07-17:
I got the same error. Have you solved it?
[https://gist.github.com/hustzxd/e0ebd676810ab52e97aee718b4ccc1ef](error)

hustzxd, 2017-07-18:
I solved the problem by change cmake version (3.5.1 to 3.9.0).
`    wget https://cmake.org/files/v3.9/cmake-3.9.0-rc5.tar.gz
     tar zxvf cmake-3.9.0-rc5.tar.gz && cd cmake-3.9.0-rc5
     ./bootstrap
     make
     sudo make install`

sria91, 2017-07-25:
@hustzxd Thanks that worked.

closed 2017-07-25 by sria91