microsoft/ELL #30: common_test failed

malynda, 2017-07-03:
While running test.py (in build/interfaces/python/test), all tests appear to pass but common_test. Built on Ubuntu 16.04.

ELL/build/interfaces/python/test$ python test.py
Testing HingeLoss.Evaluate(2, 1) ... Passed
Testing HingeLoss.Evaluate(-2, -1) ... Passed
Testing HingeLoss.Evaluate(2, -1) ... Passed
Testing HingeLoss.Evaluate(-2, 1) ... Passed
Testing HingeLoss.Evaluate(0, 1) ... Passed
Testing HingeLoss.Evaluate(0, -1) ... Passed
Testing HingeLoss.GetDerivative(2, 1) ... Passed
Testing HingeLoss.GetDerivative(-2, -1) ... Passed
Testing HingeLoss.GetDerivative(2, -1) ... Passed
Testing HingeLoss.GetDerivative(-2, 1) ... Passed
Testing HingeLoss.GetDerivative(0, 1) ... Passed
Testing HingeLoss.GetDerivative(0, -1) ... Passed
Testing LogLoss.Evaluate(2, 1) ... Passed
Testing LogLoss.Evaluate(-2, -1) ... Passed
Testing LogLoss.Evaluate(2, -1) ... Passed
Testing LogLoss.Evaluate(-2, 1) ... Passed
Testing LogLoss.Evaluate(0, 1) ... Passed
Testing LogLoss.Evaluate(0, -1) ... Passed
Testing LogLoss.GetDerivative(2, 1) ... Passed
Testing LogLoss.GetDerivative(-2, -1) ... Passed
Testing LogLoss.GetDerivative(2, -1) ... Passed
Testing LogLoss.GetDerivative(-2, 1) ... Passed
Testing LogLoss.GetDerivative(0, 1) ... Passed
Testing LogLoss.GetDerivative(0, -1) ... Passed
Testing SquaredLoss.Evaluate(4, 4) ... Passed
Testing SquaredLoss.Evaluate(4, 2) ... Passed
Testing SquaredLoss.Evaluate(2, 4) ... Passed
Testing SquaredLoss.GetDerivative(4, 4) ... Passed
Testing SquaredLoss.GetDerivative(4, 2) ... Passed
Testing SquaredLoss.GetDerivative(2, 4) ... Passed
functions_test passed
[1]	One subgraph
    Subgraph Vertices Edges Cycles
    0        6        8     3    
[2]	One subgraph
    Subgraph Vertices Edges Cycles
    0        6        6     1    
[3]	One subgraph
    Subgraph Vertices Edges Cycles
    0        8        11    4    
[tree_0]	One subgraph
    Subgraph Vertices Edges Cycles
    0        17       18    2    
[tree_1]	One subgraph
    Subgraph Vertices Edges Cycles
    0        45       54    10   
[tree_2]	One subgraph
    Subgraph Vertices Edges Cycles
    0        73       90    18   
[tree_3]	One subgraph
    Subgraph Vertices Edges Cycles
    0        101      126   26   
model_test passed
Model 1 size: 6
Model 2 size: 6
Model 3 size: 8
Tree 0 size: 17
Tree 1 size: 45
Tree 2 size: 73
Tree 3 size: 101
Loading file ../../../examples/data/model_1.model
common_test failed
trainers_test.test -- TBD
trainers_test passed
predictors_test.test -- TBD
predictors_test passed
nodes_test.test -- TBD
nodes_test passed
linear_test.test -- TBD
linear_test passed
evaluators_test.test -- TBD
evaluators_test passed
Testing ModelBuilder ... Passed
modelbuilder_test passed

lovettchris, 2017-07-06:
can you run ctest again with a -VV command line argument and post the verbose output?

lovettchris, 2017-07-15:
closing due to inactivity, please reactivate if you get an update on this, thanks.

closed 2017-07-15 by lovettchris