microsoft/ELL #33: Python: No module named 'ELL'

Fred-Erik, 2017-07-04:
I compiled ELL exactly according to the instructions [here](https://github.com/Microsoft/ELL/blob/master/interfaces/README.md) without any errors, but when I try to import ELL (or let test.py import ELL) the module cannot be found. Any idea what the problem could be? This is the output from _cmake_:
`-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libblas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libblas.so
-- Using BLAS DLLs: 
-- Found PythonInterp: /home/frederik/anaconda3/envs/ell/bin/python (found version "3.6.1") 
-- Found PythonLibs: /home/frederik/anaconda3/envs/ell/lib/libpython3.6m.so (found version "3.6.1") 
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /home/frederik/anaconda3/envs/ell/bin/python
-- Using python libraries found at: /home/frederik/anaconda3/envs/ell/lib/libpython3.6m.so
-- Creating wrappers for python
-- Configuring done
-- Generating done
-- Build files have been written to: /home/frederik/Documents/dev/experimental/ELL/build`

and this is the output from _make _ELL_python_ is also normal; the last lines are:
`[ 96%] Linking CXX static library libcommon.a
[ 96%] Built target common
Scanning dependencies of target ELL_common
[ 96%] Built target ELL_common
[ 98%] Swig source
Scanning dependencies of target _ELL_python
[ 98%] Building CXX object interfaces/python/CMakeFiles/_ELL_python.dir/ELLPYTHON_wrap.cxx.o
[100%] Building CXX object interfaces/python/CMakeFiles/_ELL_python.dir/__/common/src/ModelInterface.cpp.o
[100%] Linking CXX shared module _ELL.so
[100%] Built target _ELL_python`

But nothing is added to my conda environment in the way of libraries, nor to any other place on disk as far as I can see.

jacobmanning, 2017-07-05:
@Fred-Erik I had the same issue, were you able to run the tests (python test.py) in ELL/build/interfaces/python/test? That script is adding "ELL/build/interfaces/python" to the system path before importing ELL. It looks like the directory "ELL/build/interfaces/python" has the "_ELL.so" and "ELL.py" for importing ELL

Fred-Erik, 2017-07-06:
Thank you! The line you mentioned wasn't in my test.py, but when I added it it did manage to find _something_ at least. Is everything working for you now? Because I now get the following error:
> Traceback (most recent call last):
>   File "/home/frederik/Documents/dev/experimental/ELL/build/interfaces/python/ELL.py", line 14, in swig_import_helper
>     return importlib.import_module(mname)
>   File "/home/frederik/anaconda3/envs/ell/lib/python3.6/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 978, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 961, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 648, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 560, in module_from_spec
>   File "<frozen importlib._bootstrap_external>", line 922, in create_module
>   File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
> ImportError: /usr/lib/libblas.so.3: undefined symbol: gotoblas
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "test.py", line 14, in <module>
>     import ELL
>   File "/home/frederik/Documents/dev/experimental/ELL/build/interfaces/python/ELL.py", line 17, in <module>
>     _ELL = swig_import_helper()
>   File "/home/frederik/Documents/dev/experimental/ELL/build/interfaces/python/ELL.py", line 16, in swig_import_helper
>     return importlib.import_module('_ELL')
>   File "/home/frederik/anaconda3/envs/ell/lib/python3.6/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: /usr/lib/libblas.so.3: undefined symbol: gotoblas

I have OpenBLAS installed; the newest version, according to apt-get: "0.2.18-1ubuntu1". The files are all there in /usr/lib/

jacobmanning, 2017-07-06:
Hmm. I'm not an expert, but OpenBLAS and BLAS are two different things. It looks like that import is trying to find BLAS, not OpenBLAS. In fact, it looks like when I installed ELL cmake used BLAS instead of OpenBLAS.

My advice would be to install BLAS and LAPACK (`sudo apt-get install libblas-dev liblapack-dev`), remove the build directory (`rm -rf build`) and try to rebuild ELL (`cd build` and `cmake ..`). When you run `cmake ..` you should get a list of outputs. Make sure that "-- Blas Libraries" now points to the `/usr/lib/libblas.so` from your BLAS installation.

Also, on the topic of the original question, [in this README](https://github.com/Microsoft/ELL/blob/master/tools/importers/README.md) the instructions say to move the ELL/build/interfaces/python/ELL.py and ELL/build/interfaces/python/_ELL.so files to your Python site packages. That should solve the original ImportError once BLAS is installed.

Fred-Erik, 2017-07-07:
Thanks again! The suggestion from the README you linked solved the import issue. I already had BLAS installed but when I did `sudo apt-get remove libopenblas*` the errors stopped and test.py runs fine! So there seems to be something wrong with my OpenBLAS configuration. According to [this README](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md) you can install OpenBLAS for optimization, which I would like.

Do you have `libopenblas-dev` installed, and could you check which version? The version apt-get will installed with me is `0.2.18` (on Ubuntu 16.04) and seems not to work with ELL.

jacobmanning, 2017-07-11:
I have that same version of OpenBLAS. I haven't actually run the tutorials though so I don't know how performance is

lovettchris, 2017-07-15:
23ms on CPU is fast for darknet, which model were you using?  When I run this on my Xeon HP Z840:

`darknet -nogpu classifier predict cfg\imagenet1k.data darknet.cfg darknet.weights`

I get darknet reporting .21 seconds.  The ELL python tutorial in "reference" mode 'darknetDemo.py' gets 12 FPS (0.083 seconds) and the compiled ELL model using gets 17 fps or 0.058 seconds, and this is including the time to read from live video camera.

Fred-Erik, 2017-07-24:
Yeah, I should have mentioned that I use [DarkNet with NNPack](https://github.com/thomaspark-pkj/darknet-nnpack). Thank you for giving your performance; this means that at least for now, DarkNet is faster and gives me more options (eg. object detection, which Microsoft does not yet support) for running CNNs on ARM. But I'm curious how the development will progress!

closed 2017-07-31 by Fred-Erik