microsoft/ELL #35: Not Found Image Errors Traceback When Trying To Execute `python darknetDemo.py`

YG-FLUSH, 2017-07-04:
```
(py36) ➜  gettingStarted git:(master) python darknetDemo.py
Traceback (most recent call last):
  File "darknetDemo.py", line 4, in <module>
    import cv2
ImportError: dlopen(/Users/yoge/miniconda2/envs/py36/lib/python3.6/site-packages/cv2.cpython-36m-darwin.so, 2): Library not loaded: @rpath/libopenblasp-r0.2.19.dylib
  Referenced from: /Users/yoge/miniconda2/envs/py36/lib/libopencv_hdf.3.2.0.dylib
  Reason: image not found
```

lovettchris, 2017-07-06:
Try this [readme page on opencv](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/opencv.md).

jimaobian, 2017-07-06:
Try:
`conda update --all -c conda-forge`

YG-FLUSH, 2017-07-12:
@jimaobian thanks! it works!

closed 2017-07-15 by lovettchris