microsoft/ELL #36: compile error

liudf0716, 2017-07-05:
platform:  Ubuntu 16.04 LTS

compile result:

```
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp: In function ‘void ell::nodes::{anonymous}::EmitMatrixMatrixMultiplyBlas(ell::emitters::IRFunctionEmitter&, bool, bool, int, int, int, llvm::Value*, int, llvm::Value*, int, llvm::Value*, int)’:
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:30:34: error: ‘CBLAS_ORDER’ has not been declared
                 function.Literal(CBLAS_ORDER::CblasRowMajor), // order
                                  ^
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:31:47: error: ‘CBLAS_TRANSPOSE’ has not been declared
                 function.Literal(transposeA ? CBLAS_TRANSPOSE::CblasTrans : CBLAS_TRANSPOSE::CblasNoTrans), // transposeA
                                               ^
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:31:77: error: ‘CBLAS_TRANSPOSE’ has not been declared
                 function.Literal(transposeA ? CBLAS_TRANSPOSE::CblasTrans : CBLAS_TRANSPOSE::CblasNoTrans), // transposeA
                                                                             ^
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:32:47: error: ‘CBLAS_TRANSPOSE’ has not been declared
                 function.Literal(transposeB ? CBLAS_TRANSPOSE::CblasTrans : CBLAS_TRANSPOSE::CblasNoTrans), // transposeB
                                               ^
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:32:77: error: ‘CBLAS_TRANSPOSE’ has not been declared
                 function.Literal(transposeB ? CBLAS_TRANSPOSE::CblasTrans : CBLAS_TRANSPOSE::CblasNoTrans), // transposeB
                                                                             ^
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp: In instantiation of ‘void ell::nodes::{anonymous}::EmitMatrixMatrixMultiplyBlas(ell::emitters::IRFunctionEmitter&, bool, bool, int, int, int, llvm::Value*, int, llvm::Value*, int, llvm::Value*, int) [with ValueType = double]’:
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:98:56:   required from ‘void ell::nodes::{anonymous}::EmitMatrixMatrixMultiply(ell::emitters::IRFunctionEmitter&, bool, bool, bool, int, int, int, llvm::Value*, int, llvm::Value*, int, llvm::Value*, int) [with ValueType = double]’
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:444:52:   required from ‘void ell::nodes::DiagonalConvolutionNode<ValueType>::Compile(ell::model::IRMapCompiler&, ell::emitters::IRFunctionEmitter&) [with ValueType = double]’
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:500:1:   required from here
/home/liudf/ELL/libraries/nodes/src/ConvolutionalLayerNode.cpp:44:13: error: no matching function for call to ‘std::vector<llvm::Value*, std::allocator<llvm::Value*> >::vector(<brace-enclosed initializer list>)’
             }; // ldc
             ^

```

sambolu, 2017-07-06:
I faced this issue too when I missed installing OpenBLAS.
Please install this library and try again. It resolved this compilation issue
"sudo apt-get install -y libopenblas-dev doxygen"

liudf0716, 2017-07-06:
@sambolu thanks, I will try it.

liudf0716, 2017-07-06:
it work, thanks @sambolu

closed 2017-07-06 by liudf0716