microsoft/ELL #38: how do I get ELL.py?

ack72tdp, 2017-07-05:
I was able to compile and run unit tests successfully now ... and follow instructions 

https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/README.md

installed 
miniconda, 
OpenCV, 
numpy 
and downloaded Microsoft Cognitive Toolkit

But it seems to fail in this step now ..

root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python cntkDemo.py
Could not find './../../../build/interfaces/python/ELL.py', did you follow the ELL Python Binding build instructions?

Also, can't find compile_vggImageNet.sh ... what is ELL Python Binding build? and how can I get ELL.py file?

Thnaks in advance...

jacobrosenthal, 2017-07-05:
> vision/gettingStarted/README.md

Just another user here

did you cmake and make _ELL_python from
https://github.com/Microsoft/ELL/tree/master/interfaces

closed 2017-07-06 by ack72tdp

vyraun, 2017-08-26:
@ack72tdp @jacobrosenthal Hi, I Am getting the same error after running the vision demo:

Could not find './../../../build/interfaces/python/ELL.py', did you follow the ELL Python Binding build instructions?

Running test.py gives Import Error.

How did you resolve the issue? I Am not sure installation has to do with instructions at https://github.com/Microsoft/ELL/tree/master/interfaces.

Please take a look. Thanks.

lovettchris, 2017-08-28:
cmake --build . --config release --target _ELL_python

vyraun, 2017-08-29:
@lovettchris Thanks for the reply. However, I had executed that command and running it again says: [100%] Build target _ELL_python, while the error message "Could not find ...." continues upon running the python tutorial. I am on an Ubuntu 16.04 system. Thanks.

lovettchris, 2017-08-30:
Well, check that the build/interfaces/python/ELL.py actually exists, along with the "_ELL.pyd" native module.  I see you are using Linux so you might need "**make _ELL_python**" command line.  If these files exist and they are refusing to load this can be caused by missing openblas shared library.  But on Linux so long as you did the "sudo apt-get install -y libopenblas-dev" it should be there, you will see it in /usr/lib/libopenblas.so.  You could try running "ldd _ELL.pyd" to see if there's anything else missing.  Other than that, double checkout our [install instructions](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md).

You might also be missing "**source activate py36**" to activate your Python 3.6 environment.

This is what "ldd _ELL.so" shows on my machine:

hris@clovett14:/mnt/d/temp/ELL/linux/interfaces/python$ ldd _ELL.so
        linux-vdso.so.1 =>  (0x00007ffff6f56000)
        libpython3.6m.so.1.0 => /home/chris/miniconda3/lib/libpython3.6m.so.1.0 (0x00007f70492d9000)
        libLLVM-3.9.so.1 => /usr/lib/x86_64-linux-gnu/libLLVM-3.9.so.1 (0x00007f7046710000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f70464e0000)
        libblas.so.3 => /usr/lib/libblas.so.3 (0x00007f7046270000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f7045ee0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7045bc0000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f70459a0000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f70455d0000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f70453b0000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f70451a0000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f7044f90000)
        libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f7044d70000)
        libedit.so.2 => /usr/lib/x86_64-linux-gnu/libedit.so.2 (0x00007f7044b30000)
        libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f7044900000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f70446d0000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f704ac00000)
        libopenblas.so.0 => /usr/lib/libopenblas.so.0 (0x00007f7042630000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f7042400000)
        libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007f70420c0000)
        libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f7041e80000)