microsoft/ELL #39: SWIG missing...

ack72tdp, 2017-07-05:
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libblas.so
-- Using BLAS DLLs:
-- Found PythonInterp: /root/miniconda3/bin/python (found version "3.6.1")
-- Found PythonLibs: /root/miniconda3/lib/libpython3.6m.so (found version "3.6.1")
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
CMake Warning at CMakeLists.txt:85 (message):
  SWIG not found, interfaces and projects that depend on them will not build
  properly.  They are not part of the default make targets, so can be
  skipped.  If you want to build interfaces (such as the Python bindings),
  follow the instructions in INSTALL-XXXXX.md file to install SWIG.  Then
  delete the CMakeCache.txt file from your build directory run the cmake
  command again.

-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for python
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for javascript
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR) (Required is at least version "3.0.12")
-- Creating wrappers for xml
-- Configuring done
-- Generating done
-- Build files have been written to: /root/ELL/build

jacobrosenthal, 2017-07-05:
Just another user here. Have you tried the swig directions at
https://github.com/Microsoft/ELL/blob/244273ee64ec9432243a8d0e354d3769b642e502/interfaces/README.md

ack72tdp, 2017-07-06:
Thanks Jacob. It helped move forward bit... However I run into few other issues..

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python darknetDemo.py
Traceback (most recent call last):
  File "darknetDemo.py", line 4, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# sh compile_darknetReference.sh

Compiling darknetReference from ELL, emitting LLVM IR
Command line parse error:
Cannot read from specified input map file: darknetReference.map

Parsing darknetReference from ELL, generating SWIG interface
Command line parse error:
Cannot read from specified input map file: darknetReference.map

Compiling darknetReference LLVM IR, generating model object file
compile_darknetReference.sh: 9: compile_darknetReference.sh: llc: not found

Running disutils setup to compile and link model + SWIG wrappers into Python module, and install package locally
running build
running build_ext
building '_darknetReference' extension
swigging darknetReference.i to darknetReference_wrap.cpp
swig -python -modern -c++ -py3 -Fmicrosoft -I/root/ELL/interfaces/common -I/root/ELL/interfaces/common/include -I/root/ELL/libraries/emitters/include -o darknetReference_wrap.cpp darknetReference.i
Unable to find file 'darknetReference.i'.
error: command 'swig' failed with exit status 1

ack72tdp, 2017-07-06:
OR with microsoft cognitive kit

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python cntkDemo.py
Traceback (most recent call last):
  File "cntkDemo.py", line 4, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

jacobrosenthal, 2017-07-06:
```ModuleNotFoundError: No module named 'cv2'```
Looks like you need opencv

Thats covered here:
https://github.com/Microsoft/ELL/blob/4d3c0cc6ed13cc7885150ff74b5d3788f3ee36e8/tutorials/vision/gettingStarted/opencv.md
```conda install -c conda-forge opencv```

I actually got stuck here as well (on osx) with this issue:
```
ImportError: dlopen(/Users/jacobrosenthal/miniconda3/envs/py36/lib/python3.6/site-packages/cv2.cpython-36m-darwin.so, 2): Library not loaded: @rpath/libopenblasp-r0.2.19.dylib
  Referenced from: /Users/jacobrosenthal/miniconda3/envs/py36/lib/libopencv_hdf.3.2.0.dylib
  Reason: image not found
```

even though i installed openblas per instructions with both brew ```brew install homebrew/science/openblas
``` and conda ```conda install openblas=0.2.19```

This issue said  https://github.com/conda-forge/opencv-feedstock/issues/69
```conda update --all -c conda-forge```

Which got me up and running with ```python darknetDemo.py```

ack72tdp, 2017-07-06:
Thanks a lot Jacob. Unfortunately it didn't work in my case after running all 3 below..

conda install -c conda-forge opencv
conda install openblas=0.2.19
conda update --all -c conda-forge

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python cntkDemo.py
Traceback (most recent call last):
  File "cntkDemo.py", line 4, in <module>
    import cv2
ImportError: libGL.so.1: cannot open shared object file: No such file or directory

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python darknetDemo.py
Traceback (most recent call last):
  File "darknetDemo.py", line 4, in <module>
    import cv2
ImportError: libGL.so.1: cannot open shared object file: No such file or directory

jacobrosenthal, 2017-07-06:
Youve got a different error there you should google 
```ImportError: libGL.so.1: cannot open shared object file: No such file or directory ```

ack72tdp, 2017-07-06:
Thanks a lot Jacob. I am able to move forward much more.. Appreciate all your help.

jacobrosenthal, 2017-07-06:
Care to post your fix for others? Also if this is resolved please close your issue.

ack72tdp, 2017-07-06:
Certainly, all I did is follow step 5 ELL Python Language Bindings and Step 6.Microsoft Cognitive Toolkit install steps again and then used Step 7:Darknet model to .. run .. microsoft cognitive toolkit didn't quite work.. but darnet model seems to atleast run, but also bailed out at the end ...

Any case we can close this here and I opened new..

closed 2017-07-06 by ack72tdp