microsoft/ELL #42: implement slowly and some cfg&weights files are not work

dirksy, 2017-07-06:
ubuntu16.04 LTS
Device:Alienware 17 R4 
I7/8GB/6G(GPU)

1\shall I must copy darknet_to_ell.py to the path of darknetDemo.py?

2\if '#" in cfg file,darknetDemo.py will crash.

3\darknetDemo.py with darknet.cfg& darknet.weights run slowly
I add a timer at function main() in while loop
/××××××××××××××××××××××/
        **timer = Timer()
        timer.tic()
        # Get the model to classify the image, by returning a list of probabilities for the classes it can detect
        predictions = model.Predict(data)
        timer.toc()
        print (('Detection took {:.3f}s').format(timer.total_time))**
and display show as follow:
Detection took 1.981s
Detection took 1.993s
Detection took 1.996s
Detection took 2.021s
Detection took 1.998s
Detection took 1.990s
Detection took 1.990s

it is so slowly.where did I make a mistake? or which configuration is not correct?
(SWIG and Python version follows readme.md instructions)how can I solve this slowly problem ?if it run slowly on  my alienware laptop ，how can I try it on PI?

4\if cfg file is tiny-yolo.cfg & tiny-yolo.weights, 

**if (not os.path.exists('data/tinyyolo.cfg')):
#    if (not os.path.exists('data/darknet.cfg')):
        print("Please download the 'darknet.cfg' file, see README.md")
        sys.exit(1)
       
    if (not os.path.exists('data/tinyyolo.weights')):
#    if (not os.path.exists('data/darknet.weights')):
        print("Please download the 'darknet.weights' file, see README.md")
        sys.exit(1)

    # Pick the model you want to work with
#    helper = mh.ModelHelper("darknetReference", ["data/darknet.cfg", "data/darknet.weights"], "darknetImageNetLabels.txt")
    helper = mh.ModelHelper("darknetReference", ["data/tinyyolo.cfg", "data/tinyyolo.weights"], "data/coco.txt",(416,416))**

then darknetDemo.py crashed at modelhelpler.py's function 'get_top_n' in first cycle:
**for element in topN:
            if (element[0] > self.threshold):
                #print(element[1],int(element[1]),round(element[0],2))
                result.append((self.labels[int(element[1])], round(element[0],2)))**

error is:
**Detection took 6.795s
Traceback (most recent call last):
  File "darknetDemo.py", line 77, in <module>
    main()
  File "darknetDemo.py", line 60, in main
    top5 = helper.get_top_n(predictions, 5)
  File "/home/huimin/projects/Microsoft/ELL/build/tutorials/vision/gettingStarted/modelHelper.py", line 43, in get_top_n
    result.append((self.labels[int(element[1])], round(element[0],2)))
IndexError: list index out of range**

what is the problems?

dirksy, 2017-07-10:
could anyone tell me why my questions had nothing reply?

clovett, 2017-07-12:
Two separate questions here:
1) performance
2) using tiny-yolo.cfg & tiny-yolo.weights

On performance, you should look into the [compiling steps](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/compiling.md) which will create a faster executable.  The "darknetDemo.py" script is a model test loader that is not optimized the same way.  

Using tiny-yolo will require updating the provided darknetImageNetLabels.txt.  These labels match what is expected by darknet.cfg & darknet.weights which is probably why the self.labels[...] array index was out of range.

dirksy, 2017-07-13:
@clovett thanks for ur reply.
i have modified some codes to use new labels just as i mentioned above like this 

helper = mh.ModelHelper("darknetReference", ["data/tinyyolo.cfg", "data/tinyyolo.weights"], "data/coco.txt",(416,416))

watch param 'data/coco.txt'.
does the label name must be darknetImageNetLabels.txt? and the path must same as darknetDemo.py? i will try it soon.

btw.i have run it correctly on pi3 with darknet.cfg&weight.result is amazing and fantastic.ell just took 1300ms-1600ms for each cycle.very good

lovettchris, 2017-07-13:
I'm glad you got the compiled code working on pi3.  Congrats!

Yes it is not clear from which labels [tiny-darknet](https://pjreddie.com/darknet/tiny-darknet/) uses , I had good luck with darknetImageNetLabels.txt.  This is my modelHelper constructor:
```
helper = mh.ModelHelper(sys.argv, "darknetReference", ["yolo.cfg", "yolo.weights"], "darknetImageNetLabels.txt", inputHeightAndWidth=(224, 224))
```
Note that tiny-yolo is 224x224, not 416x416.  But I think there's a bug in our compiler because the compiled version of this model doesn't work, so I filed a bug on that already.  

[image]

dirksy, 2017-07-13:
@lovettchris thanks for your confirmation.
I am not sure '  Note that tiny-yolo is 224x224, not 416x416.  ' means. as dimension of image described in tiny-yolo.cfg&weights I used for training or testing  is 416x416. u means I can just modify the cfg file' image width&height params to 224x224? and then i am able to get correct result like your demo ? i am confused.

lovettchris, 2017-07-14:
Ah, yes, I think we are using different darknet models then, I'm using tiny.cfg with tiny.weights, as listed on this [web page](https://pjreddie.com/darknet/tiny-darknet/)  and that one is 216x216.  You are using tiny-yolo.cfg.  Where did you get that from?  I could only find tiny-yolo-**voc**.weights on this [other web page](https://pjreddie.com/darknet/yolo/) and that one is 416x416.  I found a bug in our importer with this cfg file (missing if ("=" in param): on line 39 of darknet_to_ell.py).  

The "predictions" output of that model is 13x13x125, which is 21,125 numbers, which is why the labels indices are out of range.  These are not just labels, they are 13x13 sized grids containing a label bitmask and an offset inside each grid so that bounding boxes can be computed along with those labels.  We do not have the python code in the demo yet to parse all this information.

Note also that we do not yet support the [region] layer, so the above layer output is probably not fully formed anyway.  The region detection also makes these models much larger, which is why we are only using the classification models for now.  Even the tiny-yolo-voc.weights  is twice the size of the darknet.weights that we were using.

dirksy, 2017-07-14:
thanks for your soon reply.
link of tiny-yolo.cfg :https://github.com/pjreddie/darknet/blob/master/cfg/tiny-yolo.cfg
and tiny-yolo.weights :https://pjreddie.com/media/files/tiny-yolo.weights
if you need ,I can send you label file.

all right. 
1\your first bug is in my fist comment(your parser encounter a '#"),just clean '#' in cfg file ,that will be ok.
2\will the second problem be solved ? and when ?
3\will ELL suport the third problem ? and when ?

many thanks!

lovettchris, 2017-07-14:
Yeah, ok, that one matches the tiny-yolo-voc.weights that I was using, and it does contain the region layer.  So (1) yes, scrubbing ### solves the problem.  And (2) and (3) support for [region] layers is on our backlog, I'll bring up your request in our meeting tomorrow and see what we can do for prioritizing this request :-) In the meantime if you want to take a crack at it, then all pull requests are welcome :-)

dirksy, 2017-07-14:
just waiting for ur good news.

lovettchris, 2017-07-15:
The issue is on our backlog, but it is not at the top of the list, so it will be a couple sprints out.   We have some more pressing work ahead of this.  In the meantime if you want to take a crack at it, then all pull requests are welcome :-)

closed 2017-07-15 by lovettchris