microsoft/ELL #43: darknetDemo errors

ack72tdp, 2017-07-06:
(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# python darknetDemo.py
convolutional:  224 x 224 x 3 ->  224 x 224 x 16 , pad  1
max_pool:  224 x 224 x 16 ->  112 x 112 x 16 , pad  0
convolutional:  112 x 112 x 16 ->  112 x 112 x 32 , pad  1
max_pool:  112 x 112 x 32 ->  56 x 56 x 32 , pad  0
convolutional:  56 x 56 x 32 ->  56 x 56 x 64 , pad  1
max_pool:  56 x 56 x 64 ->  28 x 28 x 64 , pad  0
convolutional:  28 x 28 x 64 ->  28 x 28 x 128 , pad  1
max_pool:  28 x 28 x 128 ->  14 x 14 x 128 , pad  0
convolutional:  14 x 14 x 128 ->  14 x 14 x 256 , pad  1
max_pool:  14 x 14 x 256 ->  7 x 7 x 256 , pad  0
convolutional:  7 x 7 x 256 ->  7 x 7 x 512 , pad  1
max_pool:  7 x 7 x 512 ->  4 x 4 x 512 , pad  1
convolutional:  4 x 4 x 512 ->  4 x 4 x 1024 , pad  1
convolutional:  4 x 4 x 1024 ->  4 x 4 x 1000 , pad  0
avg_pool:  4 x 4 x 1000 ->  1 x 1 x 1000 , pad  0
softmax:  1 x 1 x 1000 ->  1 x 1 x 1000
Traceback (most recent call last):
  File "darknetDemo.py", line 70, in <module>
    main()
  File "darknetDemo.py", line 28, in main
    model = get_ell_predictor(helper)
  File "darknetDemo.py", line 12, in get_ell_predictor
    return darknet_to_ell.predictor_from_darknet_model(helper.model_files[0], helper.model_files[1])
  File "./../../../../build/tools/importers/Darknet/darknet_to_ell.py", line 432, in predictor_from_darknet_model
    predictor = process_network(network, weightsData, convolutionOrder)
  File "./../../../../build/tools/importers/Darknet/darknet_to_ell.py", line 396, in process_network
    ellLayers += process_convolutional_layer(layer, weightsData, convolutionOrder)
  File "./../../../../build/tools/importers/Darknet/darknet_to_ell.py", line 283, in process_convolutional_layer
    weight_vals.append(struct.unpack('f', bin_data.read(4)))
struct.error: unpack requires a bytes object of length 4

(py36) root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# sh compile_darknetReference.sh
Compiling darknetReference from ELL, emitting LLVM IR
Command line parse error:
Cannot read from specified input map file: darknetReference.map
Parsing darknetReference from ELL, generating SWIG interface
Command line parse error:
Cannot read from specified input map file: darknetReference.map
Compiling darknetReference LLVM IR, generating model object file
compile_darknetReference.sh: 9: compile_darknetReference.sh: llc: not found
Running disutils setup to compile and link model + SWIG wrappers into Python module, and install package locally
running build
running build_ext
building '_darknetReference' extension
swigging darknetReference.i to darknetReference_wrap.cpp
swig -python -modern -c++ -py3 -Fmicrosoft -I/root/ELL/interfaces/common -I/root/ELL/interfaces/common/include -I/root/ELL/libraries/emitters/include -o darknetReference_wrap.cpp darknetReference.i
Unable to find file 'darknetReference.i'.
error: command 'swig' failed with exit status 1

Anyone seen these errors?

lovettchris, 2017-07-11:
The missing darknetReference.map will cause compile_darknetReference.sh to fail, so that is expected.  That will be fixed when we get darknetDemo.py to work.  Are you using the same darknet.weights and darknet.cfg listed in the darknet.md readme?

ack72tdp, 2017-07-11:
Yes, I believe so.. these are the weights and cfg downloaded and i see darnetDemo.py uses them..

curl -O https://raw.githubusercontent.com/pjreddie/darknet/master/cfg/darknet.cfg
curl -O https://pjreddie.com/media/files/darknet.weights

ack72tdp, 2017-07-11:
However, I got error when trying to get  weights file.. and so I passed it with -k option ..

oot@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# curl -O https://pjreddie.com/media/files/darknet.weights
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
root@8fa27a83d7dd:~/ELL/build/tutorials/vision/gettingStarted# curl -O https://pjreddie.com/media/files/darknet.weights -k
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  4749  100  4749    0     0   3054      0  0:00:01  0:00:01 --:--:--  3097

lovettchris, 2017-07-15:
is everything good now?

lisaong, 2017-07-26:
Hi, I'm closing out the issue since it looks like things are working.  Let us know if it isn't.

Thanks for trying ELL!

closed 2017-07-26 by lisaong