microsoft/ELL #44: Can't compile in Ubuntu 16.04

sajunkmails, 2017-07-07:
I get the following error:-
sXXX@sXXX:~/Downloads/ELL/build$ make
[  0%] Built target documentation
[ 10%] Built target utilities
[ 11%] Built target math
[ 15%] Built target data
[ 24%] Built target emitters
[ 26%] Built target evaluators
[ 29%] Built target functions
[ 39%] Built target model
[ 41%] Built target predictors
[ 51%] Built target nodes
[ 55%] Built target trainers
[ 62%] Built target common
[ 63%] Built target testing
[ 63%] Linking CXX executable common_test
libcommon.a(LoadModel.cpp.o): In function `ell::model::SteppableMap<std::chrono::_V2::system_clock>::WriteToArchive(ell::utilities::Archiver&) const':
LoadModel.cpp:(.text._ZNK3ell5model12SteppableMapINSt6chrono3_V212system_clockEE14WriteToArchiveERNS_9utilities8ArchiverE[_ZNK3ell5model12SteppableMapINSt6chrono3_V212system_clockEE14WriteToArchiveERNS_9utilities8ArchiverE]+0x7e): undefined reference to `ell::utilities::Archiver::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
LoadModel.cpp:(.text._ZNK3ell5model12SteppableMapINSt6chrono3_V212system_clockEE14WriteToArchiveERNS_9utilities8ArchiverE[_ZNK3ell5model12SteppableMapINSt6chrono3_V212system_clockEE14WriteToArchiveERNS_9utilities8ArchiverE]+0x112): undefined reference to `ell::utilities::Archiver::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
libcommon.a(LoadModel.cpp.o): In function `ell::model::SteppableMap<std::chrono::_V2::system_clock>::ReadFromArchive(ell::utilities::Unarchiver&)':
...

I think I have installed all pre-requisites.  The output of cmake is:-

-- Blas libraries: /usr/lib/libblas.so
-- Blas linker flags: 
-- Blas include directories: /usr/include
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libblas.so
-- Using BLAS DLLs: 
-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /home/sXXX/anaconda3/bin/python
-- Using python libraries found at: /home/sXXX/anaconda3/lib/libpython3.5m.so
-- Creating wrappers for python
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sXXX/Downloads/ELL/build

Thanks!

zschoyanm, 2017-07-07:
You need a newer Version of cmake and gcc.
Have the same issue with the old gcc and cmake in Ubuntu 16.04.
Use cmake 3.8.2 and gcc 6.3 and it works.

sajunkmails, 2017-07-10:
Thanks, it worked.  Managed to install the cmake and gcc versions mentioned, and passed the parameter to "make" to use gcc-6, and it worked.  Thanks again!

closed 2017-07-15 by lovettchris