microsoft/ELL #45: python version problem

sailor88128, 2017-07-07:
Python has been updated to 3.60, however, when do cmake, the error still occures....
Anybody can tell how to solve?

`pi@raspberrypi:~/Desktop/ELL/build $ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libopenblas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libopenblas.so
-- Using BLAS DLLs: 
-- Found PythonInterp: /usr/bin/python (found version "2.7.9") 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython2.7.so (found version "2.7.9") 
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
CMake Warning at interfaces/python/CMakeLists.txt:9 (message):
  Couldn't find Python libraries 3.5 or later, _ELL_python target will not be
  build properly.  This is not a required component, so it can be skipped.

-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /usr/bin/python
-- Using python libraries found at: /usr/lib/arm-linux-gnueabihf/libpython2.7.so
-- Creating wrappers for python
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/ELL/build
pi@raspberrypi:~/Desktop/ELL/build $ python3
Python 3.6.0 (default, Jul  7 2017, 19:20:32) 
[GCC 4.9.2] on linux
`

dirksy, 2017-07-07:
your python3.6 environment path must be wrong.pls check it.ell still use python2.7. see?

PeterLayton, 2017-07-08:
if you install python2.7 first, it will use python2.7 environment normally, so you can open the environment setting and add the 3.6 environment and set it front of the 2.7.

clovett, 2017-07-12:
This is why we recommend miniconda or anaconda on our [INSTALL-Ubuntu](https://github.com/Microsoft/ELL/blob/master/INSTALL-Ubuntu.md) readme instructions.  This app is cool because it can setup an "[conda environment](https://conda.io/docs/using/envs.html)" for a selected python version, then cmake will find the right version.

closed 2017-07-15 by lovettchris