microsoft/ELL #46: make failed on Ubuntu 14.04 LTS, error: undefined reference to `cblas_xxx'

codelast, 2017-07-09:
After checking the docs several times, I believe I have all the proper version of prerequisites installed on Ubuntu 14.04 LTS, but make still failed finally.
The output messages are:

> [  0%] Built target documentation
> [ 10%] Built target utilities
> [ 11%] Built target math
> [ 14%] Built target data
> [ 23%] Built target emitters
> [ 24%] Built target evaluators
> [ 28%] Built target functions
> [ 37%] Built target model
> [ 39%] Built target predictors
> [ 48%] Built target nodes
> [ 52%] Built target trainers
> [ 59%] Built target common
> [ 60%] Built target testing
> [ 61%] Linking CXX executable common_test
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, float const*, int, float*, int)':
> BlasWrapper.cpp:(.text+0x31): undefined reference to `cblas_scopy'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Copy(int, double const*, int, double*, int)':
> BlasWrapper.cpp:(.text+0x69): undefined reference to `cblas_dcopy'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Asum(int, float const*, int)':
> BlasWrapper.cpp:(.text+0x92): undefined reference to `cblas_sasum'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Asum(int, double const*, int)':
> BlasWrapper.cpp:(.text+0xba): undefined reference to `cblas_dasum'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Nrm2(int, float const*, int)':
> BlasWrapper.cpp:(.text+0xf0): undefined reference to `cblas_snrm2'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Nrm2(int, double const*, int)':
> BlasWrapper.cpp:(.text+0x118): undefined reference to `cblas_dnrm2'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Scal(int, float, float*, int)':
> BlasWrapper.cpp:(.text+0x15b): undefined reference to `cblas_sscal'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Scal(int, double, double*, int)':
> BlasWrapper.cpp:(.text+0x193): undefined reference to `cblas_dscal'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Axpy(int, float, float const*, int, float*, int)':
> BlasWrapper.cpp:(.text+0x1d9): undefined reference to `cblas_saxpy'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Axpy(int, double, double const*, int, double*, int)':
> BlasWrapper.cpp:(.text+0x221): undefined reference to `cblas_daxpy'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Dot(int, float const*, int, float const*, int)':
> BlasWrapper.cpp:(.text+0x259): undefined reference to `cblas_sdot'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Dot(int, double const*, int, double const*, int)':
> BlasWrapper.cpp:(.text+0x290): undefined reference to `cblas_ddot'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemv(CBLAS_ORDER, CBLAS_TRANSPOSE, int, int, float, float const*, int, float const*, int, float, float*, int)':
> BlasWrapper.cpp:(.text+0x309): undefined reference to `cblas_sgemv'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemv(CBLAS_ORDER, CBLAS_TRANSPOSE, int, int, double, double const*, int, double const*, int, double, double*, int)':
> BlasWrapper.cpp:(.text+0x37c): undefined reference to `cblas_dgemv'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemm(CBLAS_ORDER, CBLAS_TRANSPOSE, CBLAS_TRANSPOSE, int, int, int, float, float const*, int, float const*, int, float, float*, int)':
> BlasWrapper.cpp:(.text+0x3f5): undefined reference to `cblas_sgemm'
> ../math/libmath.a(BlasWrapper.cpp.o): In function `ell::math::Blas::Gemm(CBLAS_ORDER, CBLAS_TRANSPOSE, CBLAS_TRANSPOSE, int, int, int, double, double const*, int, double const*, int, double, double*, int)':
> BlasWrapper.cpp:(.text+0x471): undefined reference to `cblas_dgemm'
> collect2: error: ld returned 1 exit status
> make[2]: *** [libraries/common/common_test] Error 1
> make[1]: *** [libraries/common/CMakeFiles/common_test.dir/all] Error 2
> make: *** [all] Error 2

It seems that the "cblas_xxx" errors are related to openblas? But I have installed the libopenblas-dev already.
So, any suggestions? Thanks a lot.

codelast, 2017-07-09:
And below are the output of my "cmake .." command under the "build" dir:

> -- The C compiler identification is GNU 6.3.0
> -- The CXX compiler identification is GNU 6.3.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Looking for pthread.h
> -- Looking for pthread.h - found
> -- Looking for pthread_create
> -- Looking for pthread_create - not found
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
> -- Looking for pthread_create in pthread - found
> -- Found Threads: TRUE  
> -- Blas libraries: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
> -- Blas linker flags: 
> -- Blas include directories: 
> -- Using BLAS include path: /usr/include
> -- Using BLAS library: /usr/lib/libf77blas.so;/usr/lib/libatlas.so
> -- Using BLAS DLLs: 
> -- Found PythonInterp: /home/codelast/.miniconda3/bin/python (found version "3.6.1") 
> -- Found PythonLibs: /home/codelast/.miniconda3/lib/libpython3.6m.so (found version "3.6.1") 
> -- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
> -- Creating wrappers for python
> -- Creating wrappers for javascript
> -- Creating wrappers for xml
> -- Using python found at: /home/codelast/.miniconda3/bin/python
> -- Using python libraries found at: /home/codelast/.miniconda3/lib/libpython3.6m.so
> -- Creating wrappers for python
> -- Generated target compiled_vgg16ImageNet_host
> -- Generated target compiled_vgg16ImageNet_pi3
> -- Generated target compiled_vgg16ImageNet_pi0
> -- Generated target compiled_darknetReference_host
> -- Generated target compiled_darknetReference_pi3
> -- Generated target compiled_darknetReference_pi0
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/codelast/programme/pi/ELL/build

dirksy, 2017-07-09:
u have both atlas and blas . u can rename atlas files and re-cmake. then the problem solved.

codelast, 2017-07-10:
@dirksy Could you please give a more detailed info about which files to rename?
Can I just uninstall libatlas-dev?
Thank you.

dirksy, 2017-07-10:
@codelast yes.u r right.but u should make sure there r no atlas libs in path /usr/lib or usr/local/lib.

codelast, 2017-07-10:
@dirksy OK, I'll try it later, thank you.

codelast, 2017-07-10:
@dirksy After I uninstalled the following 3 packages from my Ubuntu 14.04, the "make" process finally succeeded, thank you!

> sudo apt-get remove libatlas-base-dev libatlas-dev ibatlas3-base

dirksy, 2017-07-10:
@codelast my pleasure.btw do u record demo run duration? how many ms every cycle  it took ?

closed 2017-07-12 by codelast