microsoft/ELL #47: Error at cntkDemo.py test

tdpdan, 2017-07-10:
I've worked through several issues with my build (this is my first foray into python and pi). I'm down to this error and then I think I can compile and put this baby on pi.

I get through the pooling, and then there's a whole set of "converting layer" lines that run, i noticed the word tensor on each line (this was not listed as something that happens when running the demo so it's a little confusing):

Constructing equivalent ELL layers from CNTK...
Converting layer  conv1_1: Convolution(data: Tensor[3,224,224]) -> Tensor[64,224,224]
Converting layer  relu1_1: ReLU(conv1_1: Tensor[64,224,224]) -> Tensor[64,224,224]
..(several more lines of this)

Then I get this output:

...Finished constructing ELL layers.
Traceback (most recent call last):
  File "cntkDemo.py", line 68, in <module>
    main()
  File "cntkDemo.py", line 43, in main
    data = helper.prepare_image_for_predictor(frame)
  File "C:\AIPi\ELL\build\tutorials\vision\gettingStarted\modelHelper.py", line 70, in prepare_image_for_predictor
    resized = self.resize_image(image, self.inputHeightAndWidth)
  File "C:\AIPi\ELL\build\tutorials\vision\gettingStarted\modelHelper.py", line 53, in resize_image
    if (image.shape[0] > image.shape[1]):
AttributeError: 'NoneType' object has no attribute 'shape'

Any help would be greatly appreciated!

lovettchris, 2017-07-11:
This would imply your camera is not producing images for some reason, see if you can run this little test just to view your camera output:

```
import numpy as np
import cv2
cam = cv2.VideoCapture(1)
while True:
    ret, frame = cam.read()
    cv2.imshow('test', frame)
    if cv2.waitKey(1) & 0xFF == 27:
        break
```

tdpdan, 2017-07-11:
A dark cloud of ignorance just wooshed away. Thank you lovettchris!

I did not run your test. I built this on a windows laptop that has no camera and as soon as I read your post I face-palmed.

Thank you!

closed 2017-07-11 by tdpdan

lovettchris, 2017-07-11:
no problem.  We should add better error checking in the sample and report a more useful error message if no camera is found. I filed a bug on that.