microsoft/ELL #48: python cntkDemo.py error on Ubuntu 14.04

codelast, 2017-07-10:
When running command **python cntkDemo.py** on Ubuntu 14.04, I got following errors:

> OpenBLAS : Your OS does not support AVX instructions. OpenBLAS is using Nehalem kernels as a fallback, which may give poorer performance.
> Traceback (most recent call last):
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages/cntk/cntk_py.py", line 18, in swig_import_helper
>     return importlib.import_module(mname)
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 978, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 961, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
> ModuleNotFoundError: No module named 'cntk._cntk_py'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "cntkDemo.py", line 7, in <module>
>     import cntk_to_ell
>   File "./../../../../build/tools/importers/CNTK/cntk_to_ell.py", line 17, in <module>
>     from cntk.layers import Convolution, MaxPooling, AveragePooling, Dropout, BatchNormalization, Dense
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages/cntk/__init__.py", line 10, in <module>
>     from . import cntk_py
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages/cntk/cntk_py.py", line 21, in <module>
>     _cntk_py = swig_import_helper()
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages/cntk/cntk_py.py", line 20, in swig_import_helper
>     return importlib.import_module('_cntk_py')
>   File "/home/codelast/.miniconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: libmpi_cxx.so.1: cannot open shared object file: No such file or directory

## And before that, I've successfully executed the following steps as the doc asked:
**(1) make ELL**
**(2) generate Python bindings & run the test, below are my output:**

> OpenBLAS : Your OS does not support AVX instructions. OpenBLAS is using Nehalem kernels as a fallback, which may give poorer performance.
> Testing HingeLoss.Evaluate(2, 1) ... Passed
> Testing HingeLoss.Evaluate(-2, -1) ... Passed
> Testing HingeLoss.Evaluate(2, -1) ... Passed
> Testing HingeLoss.Evaluate(-2, 1) ... Passed
> Testing HingeLoss.Evaluate(0, 1) ... Passed
> Testing HingeLoss.Evaluate(0, -1) ... Passed
> Testing HingeLoss.GetDerivative(2, 1) ... Passed
> Testing HingeLoss.GetDerivative(-2, -1) ... Passed
> Testing HingeLoss.GetDerivative(2, -1) ... Passed
> Testing HingeLoss.GetDerivative(-2, 1) ... Passed
> Testing HingeLoss.GetDerivative(0, 1) ... Passed
> Testing HingeLoss.GetDerivative(0, -1) ... Passed
> Testing LogLoss.Evaluate(2, 1) ... Passed
> Testing LogLoss.Evaluate(-2, -1) ... Passed
> Testing LogLoss.Evaluate(2, -1) ... Passed
> Testing LogLoss.Evaluate(-2, 1) ... Passed
> Testing LogLoss.Evaluate(0, 1) ... Passed
> Testing LogLoss.Evaluate(0, -1) ... Passed
> Testing LogLoss.GetDerivative(2, 1) ... Passed
> Testing LogLoss.GetDerivative(-2, -1) ... Passed
> Testing LogLoss.GetDerivative(2, -1) ... Passed
> Testing LogLoss.GetDerivative(-2, 1) ... Passed
> Testing LogLoss.GetDerivative(0, 1) ... Passed
> Testing LogLoss.GetDerivative(0, -1) ... Passed
> Testing SquaredLoss.Evaluate(4, 4) ... Passed
> Testing SquaredLoss.Evaluate(4, 2) ... Passed
> Testing SquaredLoss.Evaluate(2, 4) ... Passed
> Testing SquaredLoss.GetDerivative(4, 4) ... Passed
> Testing SquaredLoss.GetDerivative(4, 2) ... Passed
> Testing SquaredLoss.GetDerivative(2, 4) ... Passed
> functions_test passed
> [1] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        6        8     3    
> [2] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        6        6     1    
> [3] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        8        11    4    
> [tree_0] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        17       18    2    
> [tree_1] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        45       54    10   
> [tree_2] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        73       90    18   
> [tree_3] One subgraph
>     Subgraph Vertices Edges Cycles
>     0        101      126   26   
> model_test passed
> Model 1 size: 6
> Model 2 size: 6
> Model 3 size: 8
> Tree 0 size: 17
> Tree 1 size: 45
> Tree 2 size: 73
> Tree 3 size: 101
> Loading file ../../../examples/data/model_1.model
> Loading file ../../../examples/data/model_2.model
> common_test passed
> trainers_test.test -- TBD
> trainers_test passed
> predictors_test.test -- TBD
> predictors_test passed
> nodes_test.test -- TBD
> nodes_test passed
> linear_test.test -- TBD
> linear_test passed
> evaluators_test.test -- TBD
> evaluators_test passed
> Testing ModelBuilder ... Passed
> modelbuilder_test passed

**(3) install CNTK python packages:**

> (py36) [codelast@ gettingStarted]$ pip install https://cntk.ai/PythonWheel/CPU-Only/cntk-2.0-cp36-cp36m-linux_x86_64.whl
> Requirement already satisfied: cntk==2.0 from https://cntk.ai/PythonWheel/CPU-Only/cntk-2.0-cp36-cp36m-linux_x86_64.whl in /home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages
> Requirement already satisfied: numpy>=1.11 in /home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages (from cntk==2.0)
> Requirement already satisfied: scipy>=0.17 in /home/codelast/.miniconda3/envs/py36/lib/python3.6/site-packages (from cntk==2.0)

**(4) download the CNTK ImageNet reference model**

> curl -O https://www.cntk.ai/Models/Caffe_Converted/VGG16_ImageNet_Caffe.model

So, could anyone give me a hint on how to solve the problem? Thanks a lot.

lovettchris, 2017-07-11:
_> ImportError: libmpi_cxx.so.1: cannot open shared object file: No such file or directory_

I wonder if "sudo apt-get install libopenmpi-dev" will solve this problem?

There's a lot more [CNTK Linux instructions](https://docs.microsoft.com/en-us/cognitive-toolkit/Setup-CNTK-on-Linux#open-mpi) if that doesn't work, one would assume the conda cntk python package should have taken care of all these prerequisites, but perhaps it has a bug...

codelast, 2017-07-11:
@lovettchris Thank you for providing the info, I'll try it later.

codelast, 2017-07-11:
@lovettchris 
sudo apt-get install libopenmpi-dev can't solve the problem because on Ubuntu 14.04 the package installed by this way is too old:

> [codelast@ ~]$ apt-cache show libopenmpi-dev
> Version: 1.6.5-8
> 
> [codelast@ ~]$ ll /usr/lib/libmpi.so*
> lrwxrwxrwx 1 root root 27  7 12 00:03 /usr/lib/libmpi.so -> /etc/alternatives/libmpi.so
> lrwxrwxrwx 1 root root 15 12 29  2013 /usr/lib/libmpi.so.1 -> libmpi.so.1.0.8
> lrwxrwxrwx 1 root root 27 12 29  2013 /usr/lib/libmpi.so.1.0.8 -> openmpi/lib/libmpi.so.1.0.8

## While ELL requires libmpi.so.12, so I have to do the following things:
**(1) uninstall all the openmpi-related packages on my Ubuntu 14.04:**

> sudo apt-get remove libmpich-dev mpi-default-dev libopenmpi-dev libopenmpi1.6 openmpi-common

**(2) download the openmpi-1.10.2.tar.gz package on [this](https://www.open-mpi.org/software/ompi/v1.10/) web page, and then :**

> tar zxf openmpi-1.10.2.tar.gz
> cd openmpi-1.10.2/
> ./configure
> make
> sudo make install

**(3) check the installation:**

> [codelast@ ~]$ sudo find ./ -name libmpi.so.12
> ./usr/local/lib/libmpi.so.12

That's it! Problem solved!

closed 2017-07-13 by codelast