microsoft/ELL #50: question:can't make in Ubuntu 14.04

devilloser, 2017-07-11:
IRExecutionEngine.cpp:(.text+0x532): undefined reference to `llvm::EngineBuilder::selectTarget(llvm::Triple const&, llvm::StringRef, llvm::StringRef, llvm::SmallVectorImpl<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)'
libemitters.a(IRAssemblyWriter.cpp.o): In function `ell::emitters::GenerateMachineCode(llvm::raw_ostream&, ell::emitters::IRModuleEmitter&, llvm::TargetMachine::CodeGenFileType, ell::emitters::MachineCodeOutputOptions const&)':
IRAssemblyWriter.cpp:(.text+0x4d7): undefined reference to `llvm::sys::getDefaultTargetTriple[abi:cxx11]()'
IRAssemblyWriter.cpp:(.text+0x542): undefined reference to `llvm::Triple::normalize[abi:cxx11](llvm::StringRef)'
IRAssemblyWriter.cpp:(.text+0x607): undefined reference to `llvm::TargetRegistry::lookupTarget(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, llvm::Triple&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [libraries/emitters/emitters_test] Error 1
make[1]: *** [libraries/emitters/CMakeFiles/emitters_test.dir/all] Error 2
[ 75%] Linking CXX static library libcommon.a
[ 75%] Built target common
make: *** [all] Error 2
Is there something wrong with LLVM？

lisaong, 2017-07-12:
Hi,
It's hard to tell from the error, maybe you're using the wrong version of LLVM. Can you paste the output of `cmake ..`?  We've been developing with LLVM 3.9 and 3.9.1 because that's what is more widely available.

devilloser, 2017-07-13:
@lisaong 
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/local/bin/gcc
-- Check for working C compiler: /usr/local/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/bin/c++
-- Check for working CXX compiler: /usr/local/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libopenblas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libopenblas.so
-- Using BLAS DLLs: 
-- Found PythonInterp: /usr/bin/python (found version "3.6") 
-- Found PythonLibs: /usr/local/lib/libpython3.6m.a (found version "3.6.0") 
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
-- Creating wrappers for python
CMake Deprecation Warning at /home/zf/develop/cmake-3.8.2-Linux-x86_64/share/cmake-3.8/Modules/UseSWIG.cmake:226 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:159 (swig_add_module)
  CMake/CommonInterfaces.cmake:190 (generate_interface_module)
  interfaces/python/CMakeLists.txt:21 (generate_interface)

-- Creating wrappers for javascript
-- Found npm at /usr/bin/npm
-- Creating wrappers for xml
-- Using python found at: /usr/bin/python
-- Using python libraries found at: /usr/local/lib/libpython3.6m.a
-- Creating wrappers for python
CMake Deprecation Warning at /home/zf/develop/cmake-3.8.2-Linux-x86_64/share/cmake-3.8/Modules/UseSWIG.cmake:226 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  CMake/CommonInterfaces.cmake:159 (swig_add_module)
  CMake/CommonInterfaces.cmake:237 (generate_interface_module)
  examples/emitted_interfaces/step/CMakeLists.txt:25 (generate_emitted_interface_module)

-- Generated target compiled_vgg16ImageNet_host
-- Generated target compiled_vgg16ImageNet_pi3
-- Generated target compiled_vgg16ImageNet_pi0
-- Generated target compiled_darknetReference_host
-- Generated target compiled_darknetReference_pi3
-- Generated target compiled_darknetReference_pi0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zf/ELL/build

**this is the output of cmake ..，and my LLVM version is 3.9.1**

lisaong, 2017-07-13:
Hmm...Maybe your LLVM libs are not setup correctly.

Check what llvm libs are being linked against in link.txt under build/libraries/emitters/CMakeFiles/emitters_test.dir.

Once you find the libs, you can use nm to dump the libs and grep for getDefaultTargetTriple. If that doesn't exist, try reinstalling llvm...

closed 2017-07-19 by devilloser