microsoft/ELL #52: Docker running in Rasberry PI or PC..

ack72tdp, 2017-07-11:
Does any one tried to make it work inside docker on a rasberry pi or PC? 

I was able to compile and run all unit tests and to some what darnetDemo.py file .. But not clear on how to use camera connected to PC or Rasberry PI.. and hooking it up with docker linux running ELL..

ack72tdp, 2017-07-11:
Or if any one has docker image with ELL compiled and ready to go ... it would be great if you publish that image. 

Also, instructions on how to hook up camera in PC or Rasberry PI to dockerized ELL ..

DarrenRainey, 2017-07-26:
I havent used docker that much but this might help : https://stackoverflow.com/questions/34302096/sharing-devices-webcam-usb-drives-etc-with-docker

also guess you could try streaming your camera over the network to the docker machine if you can't get the hardware pass though working.

lisaong, 2018-05-17:
Here's a Raspberry Pi 3 docker with ELL and docker:

https://hub.docker.com/r/lisaong/pi3-miniconda3-ell

To run it and mount a folder containing your Jupyter notebooks:

```
docker run -it \
  -v your_notebook_location:/notebooks/myNotebooks \
  -v /opt/vc:/opt/vc \
  -p:28888:8888 \
  lisaong/pi3-miniconda3-ell:2.3.3
```

Note that this container is designed to run on a Raspberry Pi 3.  To install Docker on Raspberry Pi 3:

```
curl -sSL get.docker.com | sh
sudo usermod -aG docker pi
```

log out, then log back in again for the change to take effect

```
sudo systemctl start docker
```

Reference: https://github.com/romilly/rpi-docker-tensorflow

Hope this helps,
lisa

grahamehorner, 2018-12-18:
You could try https://hub.docker.com/r/ryankurte/ this is an emulator of the Raspberry PI environment running inside docker