microsoft/ELL #53: python cntkDemo.py error: OpenCV Error: Unspecified error in cvShowImage

codelast, 2017-07-12:
I was able to successfully finish all the steps before running python cntkDemo.py on my Ubuntu 14.04, and failed after a long time running python cntkDemo.py:

> OpenBLAS : Your OS does not support AVX instructions. OpenBLAS is using Nehalem kernels as a fallback, which may give poorer performance.
> Loading...
> Selected CPU as the process wide default device.
>  
> Finished loading.
> Pre-processing...
>  
> Will not process Dropout - skipping this layer as irrelevant.
> Will not process Dropout - skipping this layer as irrelevant.
> Will not process Combine - skipping this layer as irrelevant.
> Convolution :  226x226x3  ->  224x224x64 | padding  1
> ReLU :  224x224x64  ->  226x226x64 | padding  0
> ......
> Softmax :  1x1x1000  ->  1x1x1000 | padding  0
>  
> Finished pre-processing.
>  
> Constructing equivalent ELL layers from CNTK...
> Converting layer  conv1_1: Convolution(data: Tensor[3,224,224]) -> Tensor[64,224,224]
> ......
> Converting layer  prob: Softmax(fc8: Tensor[1000]) -> Tensor[1000]
>  
> ...Finished constructing ELL layers.
> OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvShowImage, file /feedstock_root/build_artefacts/opencv_1490907195496/work/opencv-3.2.0/modules/highgui/src/window.cpp, line 583
> Traceback (most recent call last):
>   File "cntkDemo.py", line 68, in <module>
>     main()
>   File "cntkDemo.py", line 61, in main
>     cv2.imshow('frame', frameToShow)
> cv2.error: /feedstock_root/build_artefacts/opencv_1490907195496/work/opencv-3.2.0/modules/highgui/src/window.cpp:583: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvShowImage

It seems that the error occurred when the program tried to pop up a window on Ubuntu.
I followed the instruction in the error message(install libgtk2.0-dev and pkg-config, and re-cmake, re-make ELL), but the problem still exists.

Hope I can get some help from anyone, thanks a lot.

clovett, 2017-07-12:
Can you run this opencv test to see if your camera works?

```
import numpy as np
import cv2
cap = cv2.VideoCapture(0)
while True:
    ret, frame = cap.read()
    if (ret == False):
        print("Error reading from camera")
        break
    cv2.imshow('frame',frame)
    key = cv2.waitKey(1) & 0xFF;
    if (key == 27 or key == ord('q') or key == ord('x')):
        break;

cap.release()
cv2.destroyAllWindows()
```

codelast, 2017-07-12:
@clovett I tried, and it didn't work.
After hours of search I finally figured it out why.
On my Ubuntu 14.04, the way by [ELL doc](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/opencv.md) to install opencv caused the error result:

> conda install -c conda-forge opencv

Here is the solution:

> conda install --channel loopbio --channel conda-forge --channel pkgw-forge gtk2 ffmpeg ffmpeg-feature gtk2-feature opencv

Refer to the following two links for details:
(1) https://stackoverflow.com/questions/40207011/opencv-not-working-properly-with-python-on-linux-with-anaconda-getting-error-th
(2) https://github.com/conda-forge/opencv-feedstock/issues/43

closed 2017-07-13 by codelast