microsoft/ELL #54: make failed on Raspberry Pi 3B, Raspbian ARMv7

cocteau666, 2017-07-12:
Hi, I'm trying to build an ELL binary on Raspbian ARMv7.
But I got few errors like below.

`[ 44%] Building CXX object libraries/nodes/CMakeFiles/nodes.dir/src/BinaryConvolutionalLayerNode.cpp.o
In file included from /root/ELL/libraries/utilities/include/Archiver.h:430:0,
                 from /root/ELL/libraries/utilities/include/IArchivable.h:11,
                 from /root/ELL/libraries/utilities/include/UniqueId.h:11,
                 from /root/ELL/libraries/model/include/Node.h:11,
                 from /root/ELL/libraries/model/include/Model.h:11,
                 from /root/ELL/libraries/model/include/IRModelProfiler.h:11,
                 from /root/ELL/libraries/model/include/CompilableNode.h:11,
                 from /root/ELL/libraries/nodes/include/NeuralNetworkLayerNode.h:12,
                 from /root/ELL/libraries/nodes/include/BinaryConvolutionalLayerNode.h:11,
                 from /root/ELL/libraries/nodes/src/BinaryConvolutionalLayerNode.cpp:9:
/root/ELL/libraries/utilities/include/../tcc/Archiver.tcc: In instantiation of ‘void ell::utilities::Archiver::ArchiveItem(const char*, const std::vector<T>&) [with ValueType = long long unsigned int; typename std::enable_if<std::is_fundamental<typename std::decay<_Tp>::type>::value, int>::type concept = 0]’:
/root/ELL/libraries/utilities/include/../tcc/Archiver.tcc:42:9:   required from ‘void ell::utilities::Archiver::Archive(const char*, ValueType&&) [with ValueType = std::vector<long long unsigned int, std::allocator<long long unsigned int> >&]’
/root/ELL/libraries/utilities/include/../tcc/Archiver.tcc:20:9:   required from ‘void ell::utilities::Archiver::PropertyArchiver::operator<<(ValueType&&) [with ValueType = std::vector<long long unsigned int, std::allocator<long long unsigned int> >&]’
/root/ELL/libraries/predictors/neural/include/../tcc/BinaryConvolutionalLayer.tcc:294:45:   required from ‘void ell::predictors::neural::BinaryConvolutionalLayer<ElementType>::WriteToArchive(ell::utilities::Archiver&) const [with ElementType = double]’
/root/ELL/libraries/nodes/src/BinaryConvolutionalLayerNode.cpp:549:1:   required from here
/root/ELL/libraries/utilities/include/../tcc/Archiver.tcc:67:9: error: no matching function for call to ‘ell::utilities::Archiver::ArchiveArray(const char*&, const std::vector<long long unsigned int, std::allocator<long long unsigned int> >&)’
         ArchiveArray(name, array);
         ^~~~~~~~~~~~
In file included from /root/ELL/libraries/utilities/include/IArchivable.h:11:0,
                 from /root/ELL/libraries/utilities/include/UniqueId.h:11,
                 from /root/ELL/libraries/model/include/Node.h:11,
                 from /root/ELL/libraries/model/include/Model.h:11,
                 from /root/ELL/libraries/model/include/IRModelProfiler.h:11,
                 from /root/ELL/libraries/model/include/CompilableNode.h:11,
                 from /root/ELL/libraries/nodes/include/NeuralNetworkLayerNode.h:12,
                 from /root/ELL/libraries/nodes/include/BinaryConvolutionalLayerNode.h:11,
                 from /root/ELL/libraries/nodes/src/BinaryConvolutionalLayerNode.cpp:9:
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note: candidate: virtual void ell::utilities::Archiver::ArchiveArray(const char*, const std::vector<bool>&, ell::utilities::IsFundamental<bool>)
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:200:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(bool);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note:   no known conversion for argument 2 from ‘const std::vector<long long unsigned int, std::allocator<long long unsigned int> >’ to ‘const std::vector<bool>&’
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:200:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(bool);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note: candidate: virtual void ell::utilities::Archiver::ArchiveArray(const char*, const std::vector<char>&, ell::utilities::IsFundamental<char>)
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:201:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(char);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note:   no known conversion for argument 2 from ‘const std::vector<long long unsigned int, std::allocator<long long unsigned int> >’ to ‘const std::vector<char>&’
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:201:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(char);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note: candidate: virtual void ell::utilities::Archiver::ArchiveArray(const char*, const std::vector<short int>&, ell::utilities::IsFundamental<short int>)
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:202:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(short);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note:   no known conversion for argument 2 from ‘const std::vector<long long unsigned int, std::allocator<long long unsigned int> >’ to ‘const std::vector<short int>&’
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
/root/ELL/libraries/utilities/include/Archiver.h:202:9: note: in expansion of macro ‘DECLARE_ARCHIVE_ARRAY_BASE’
         DECLARE_ARCHIVE_ARRAY_BASE(short);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:69:55: note: candidate: virtual void ell::utilities::Archiver::ArchiveArray(const char*, const std::vector<int>&, ell::utilities::IsFundamental<int>)
 #define DECLARE_ARCHIVE_ARRAY_BASE(type) virtual void ArchiveArray(const char* name, const std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                       ^
[... 173 lines cut ...]
/root/ELL/libraries/utilities/include/Archiver.h:336:9: note: in expansion of macro ‘DECLARE_UNARCHIVE_ARRAY_BASE’
         DECLARE_UNARCHIVE_ARRAY_BASE(float);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:241:57: note: candidate: virtual void ell::utilities::Unarchiver::UnarchiveArray(const char*, std::vector<double>&, ell::utilities::IsFundamental<double>)
 #define DECLARE_UNARCHIVE_ARRAY_BASE(type) virtual void UnarchiveArray(const char* name, std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
                                                         ^
/root/ELL/libraries/utilities/include/Archiver.h:337:9: note: in expansion of macro ‘DECLARE_UNARCHIVE_ARRAY_BASE’
         DECLARE_UNARCHIVE_ARRAY_BASE(double);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ELL/libraries/utilities/include/Archiver.h:241:57: note:   no known conversion for argument 2 from ‘std::vector<long long unsigned int, std::allocator<long long unsigned int> >’ to ‘std::vector<double>&’
 #define DECLARE_UNARCHIVE_ARRAY_BASE(type) virtual void UnarchiveArray(const char* name, std::vector<type>& value, IsFundamental<type> dummy = 0) = 0;
[... cut ...] _string<char> >&’
libraries/nodes/CMakeFiles/nodes.dir/build.make:134: recipe for target 'libraries/nodes/CMakeFiles/nodes.dir/src/BinaryConvolutionalLayerNode.cpp.o' failed
make[2]: *** [libraries/nodes/CMakeFiles/nodes.dir/src/BinaryConvolutionalLayerNode.cpp.o] Error 1
CMakeFiles/Makefile2:1032: recipe for target 'libraries/nodes/CMakeFiles/nodes.dir/all' failed
make[1]: *** [libraries/nodes/CMakeFiles/nodes.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2`

cocteau666, 2017-07-12:
My cmake result is below.

`# cmake .. -DPYTHON_INCLUDE_DIR=/usr/include/python3.4m -DPYTHON_LIBRARY=/usr/lib/arm-linux-gnueabihf/libpython3.4m.so -DPYTHON_EXECUTABLE=/usr/bin/python3
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Blas libraries: /usr/lib/libopenblas.so
-- Blas linker flags: 
-- Blas include directories: 
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libopenblas.so
-- Using BLAS DLLs: 
-- Found PythonInterp: /usr/bin/python3 (found version "3.4.2") 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython3.4m.so  
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.12", minimum required is "3.0.12") 
-- Creating wrappers for python
-- Creating wrappers for javascript
-- Creating wrappers for xml
-- Using python found at: /usr/bin/python3
-- Using python libraries found at: /usr/lib/arm-linux-gnueabihf/libpython3.4m.so
-- Creating wrappers for python
-- Generated target compiled_vgg16ImageNet_host
-- Generated target compiled_vgg16ImageNet_host_profile
-- Generated target compiled_vgg16ImageNet_pi3
-- Generated target compiled_vgg16ImageNet_pi3_profile
-- Generated target compiled_vgg16ImageNet_pi0
-- Generated target compiled_vgg16ImageNet_aarch64
-- Generated target compiled_darknetReference_host
-- Generated target compiled_darknetReference_host_profile
-- Generated target compiled_darknetReference_pi3
-- Generated target compiled_darknetReference_pi3_profile
-- Generated target compiled_darknetReference_pi0
-- Generated target compiled_darknetReference_aarch64
-- Configuring done
-- Generating done
-- Build files have been written to: /root/ELL/build`

SWIG has been compiled from source code.
gcc-6 has been installed from stretch repository.

lisaong, 2017-07-12:
See: https://github.com/Microsoft/ELL/issues/14

closed 2017-07-12 by lisaong