microsoft/ELL #55: How can I use ELL with C++ or C?

alan-gu, 2017-07-18:
Dear ELL developers,
I'm looking for a machine learning library for my embedded system(1.5M RAM) running Nuttx.
I read ELL tutorial and built compiled_darknetReference_aarch64. But it is wrapped with python which will not work on the very primitive environment of Nuttx.
Can you give me some ideas to use it with C or C++?
Thank you.

lovettchris, 2017-07-18:
Sure thing, you don't have to use the python wrappers, you will notice in the generated header file darknetReference_i.h which contains this function:
```
void darknetReference_predict(float*, float*);
```
Just call it with the correct size input and output buffers and you will get your results in C++.  The correct size for these buffers is provided in the same header file via darknetReference_GetInputSize and darknetReference_GetOutputSize.

lisaong, 2017-07-26:
Hi @alan-gu, let us know if this doesn't works for you.  We'll close out the issue in the meantime.

closed 2017-07-26 by lisaong

alan-gu, 2017-08-03:
Sorry for the delay.
Thank you for your answer. I managed to run the program, but the result was undesirable. The worst thing is that I just know too little about machine learning to have any thought on this.
Our group is completely new to this machine learning thing. After trying for about a week, we decided to get more general ideas about machine learning before applying it to our system, which seems to be a very advanced  task. We hope we could be able to challenge this task in 3 months.
Thank you again.

xhMatt, 2017-08-15:
Thanks, I compiled the source code without error, but I cant find the generated header file darknetReference_i.h in ELL directory or any of subdirectory (Ubuntu 16.04),  what I am trying to do is translate your *.py to *.cpp, but I don't think this could help, thanks!