microsoft/ELL #57: failed to build the compiled CNTK model on Rasberry Pi 3

codelast, 2017-07-20:
When builing the compiled CNTK model on Rasberry Pi 3 by [this tutorial](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/compiling-Pi3.md), I failed to finish the make process.

Here are my steps:
(1) scp the compiled CNTK model to Raspberry Pi 3, unpack it on Pi
(2) cmake under the miniconda environment, output messages are:
> -- (py34)[root@alarmpi build]# cmake ..
> -- The C compiler identification is GNU 7.1.1       
> -- The CXX compiler identification is GNU 7.1.1     
> -- Check for working C compiler: /usr/bin/cc        
> -- Check for working C compiler: /usr/bin/cc -- works                                                   
> -- Detecting C compiler ABI info                    
> -- Detecting C compiler ABI info - done             
> -- Detecting C compile features                     
> -- Detecting C compile features - done              
> -- Check for working CXX compiler: /usr/bin/c++     
> -- Check for working CXX compiler: /usr/bin/c++ -- works                                                
> -- Detecting CXX compiler ABI info                  
> -- Detecting CXX compiler ABI info - done           
> -- Detecting CXX compile features                   
> -- Detecting CXX compile features - done            
> -- Looking for pthread.h  
> -- Looking for pthread.h - found                    
> -- Looking for pthread_create                       
> -- Looking for pthread_create - not found           
> -- Looking for pthread_create in pthreads           
> -- Looking for pthread_create in pthreads - not found                                                   
> -- Looking for pthread_create in pthread            
> -- Looking for pthread_create in pthread - found    
> -- Found Threads: TRUE    
> -- Blas libraries: /usr/lib/libblas.so              
> -- Blas linker flags:     
> -- Blas include directories:                        
> -- Using BLAS include path: /usr/include            
> -- Using BLAS library: /usr/lib/libblas.so          
> -- Using BLAS DLLs:       
> -- Found PythonInterp: /root/.miniconda3/envs/py34/bin/python3.4 (found suitable version "3.4.3", minimum required is "3.4") 
> -- Found PythonLibs: /root/.miniconda3/envs/py34/lib/libpython3.4m.so (found suitable version "3.4.3", minimum required is "3.4") 
> -- Configuring done       
> -- Generating done        
> -- Build files have been written to: /root/raspberry-pi/ai/ell-related/compiled_vgg16ImageNet_pi3/build

(3) make and then failed, part of the output messages are:

> Scanning dependencies of target _vgg16ImageNet                                                                                                               
> [ 50%] Building CXX object CMakeFiles/_vgg16ImageNet.dir/vgg16ImageNetPYTHON_wrap.cxx.o
> ......
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h: In function ‘typename std::iterator_traits<_Iterator>::difference_type std::__distance(_RandomAccessIterator, _RandomAccessIterator, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >]’:
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1
>      __distance(_RandomAccessIterator __first, _RandomAccessIterator __last,                            
>      ^~~~~~~~~~
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h: In function ‘typename std::iterator_traits<_Iterator>::difference_type std::__distance(_RandomAccessIterator, _RandomAccessIterator, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >]’:
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1
>      __distance(_RandomAccessIterator __first, _RandomAccessIterator __last,                            
>      ^~~~~~~~~~
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1/usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h: In function ‘typename std::iterator_traits<_Iterator>::difference_type std::__distance(_RandomAccessIterator, _RandomAccessIterator, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >]’:
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1
>      __distance(_RandomAccessIterator __first, _RandomAccessIterator __last,                            
>      ^~~~~~~~~~
> /usr/include/c++/7.1.1/bits/stl_iterator_base_funcs.h:98:5: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::vector<double, std::allocator<double> > >’ changed in GCC 7.1
> [100%] Linking CXX shared library _vgg16ImageNet.so 
> ../vgg16ImageNet.o: file not recognized: File truncated                                                 
> collect2: error: ld returned 1 exit status          
> make[2]: *** [CMakeFiles/_vgg16ImageNet.dir/build.make:99: _vgg16ImageNet.so] Error 1                   
> make[1]: *** [CMakeFiles/Makefile2:68: CMakeFiles/_vgg16ImageNet.dir/all] Error 2                       
> make: *** [Makefile:84: all] Error 2

BTW, the OS of my Raspberry Pi 3 is:
Arch Linux ARM
(Linux version 4.9.36-1-ARCH (builduser@leming) (gcc version 7.1.1 20170516 (GCC) ) # 1 SMP Fri Jul 7 02:16:47 UTC 2017)

Does anyone know how to solve this problem? Thanks a lot.

lovettchris, 2017-07-21:
are you actually using GCC 7.1 on a Pi ? I've never done that.  I've been using GCC 5.4 and 6 and ELL compiles fine using those versions. You'll need LLVM on there too, if it fits.  How did you do that?  I normally build the raspberry pi binaries on my host PC then download just the compiled binaries to the pi for running there.  Then I don't need the full ELL setup on the Pi itself.  This process is described in the [tutorials](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/README.md).

codelast, 2017-07-22:
@lovettchris 
Maybe it's not very common to install Arch Linux ARM on RPi comparing to Raspbian, and Arch Linux ARM for RPi has a very new gcc version(under default installation), as you seen, 7.1.1.
It's not easy to install gcc 6 on Arch Linux ARM - I tried to build gcc 6 from source, and it took a very very long time & failed finally :(

I've read the tutorial in your post, and if I did't misunderstand it, the problem I mentioned in this issue is in the step "Compiling the model code on target device into a Python module", and what you said "I normally build the raspberry pi binaries on my host PC" must be under the condition that your cross-compilation toolchain properly setup on your PC, right?

So I'm in a embarrassing situation: firstly I have no cross-compilation toolchain on my Ubuntu PC, secondly my RPi OS has gcc 7.1.1 which may too new.

I consider maybe reinstalling Raspbian on my RPi a more time-saving way to solve my problem?
Any other suggestions?
Thank you.

lovettchris, 2017-07-22:
Ah, I wasn't clear, the host PC can use LLVM to cross-compile the "darknetReference.so" library, this is why we need LLVM, it already has cross-compilation support.  So if you do have LLVM on your host PC then you have everything you need.  Then the "darknetReference.so" library is used in a little cmake project to build the final binaries on the Pi itself.  So this way the Pi doesn't need to build all of ELL, just the tutorial.  When you run `cmake --build . --config Release --target compiled_darknetReference_pi3` you will notice that it creates the  "darknetReference.so" library already targeted for the pi.  You then copy compiled_darknetReference_pi3 to the pi and finish the compile/link there.  If you are using a 64bit arch Linux install on the Pi then you might need to use  compiled_darknetReference_aarch64 instead.

codelast, 2017-07-23:
I did try to build the target .so several times on my Ubuntu PC, but all of them failed and didn't provide a clear error reason:

> (py36) [codelast@ build]$ make compiled_vgg16ImageNet_pi3
> [ 16%] Built target utilities
> [ 18%] Built target math
> [ 23%] Built target data
> [ 38%] Built target emitters
> [ 40%] Built target evaluators
> [ 45%] Built target functions
> [ 60%] Built target model
> [ 63%] Built target predictors
> [ 78%] Built target nodes
> [ 85%] Built target trainers
> [ 96%] Built target common
> [ 98%] Built target compile
> [ 98%] Generating /home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.ll;/home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.i;/home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.i.h
> OpenBLAS : Your OS does not support AVX instructions. OpenBLAS is using Nehalem kernels as a fallback, which may give poorer performance.
> [100%] Compiling vgg16ImageNet.ll to /home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.o for pi3
> Killed
> make[3]: *** [tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.o] Error 137
> make[3]: *** Deleting file `tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.o'
> make[2]: *** [tutorials/vision/gettingStarted/CMakeFiles/compiled_vgg16ImageNet_pi3.dir/all] Error 2
> make[1]: *** [tutorials/vision/gettingStarted/CMakeFiles/compiled_vgg16ImageNet_pi3.dir/rule] Error 2
> make: *** [compiled_vgg16ImageNet_pi3] Error 2

I don't know whether it has more detailed log about the error ?

lisaong, 2017-07-26:
Hi,

Compared to the darknetReference model, the vgg16 model is much larger.  As a result, compiling it on the PC (regardless of target) will take quite a while.

```
[100%] Compiling vgg16ImageNet.ll to /home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.o for pi3
Killed
````
What the error message shows above is probably your compiler taking a very long time (which is expected for vgg).  

We are aware of this limitation of off-the-shelf models, and working on coming up with smaller models that work better for embedded devices.  For now, I recommend sticking to darknetReference, which is a smaller and more manageable size.

codelast, 2017-07-26:
@lisaong Thank you so much & I'll try darknet later.

DarrenRainey, 2017-07-26:
Looking through your error log I've found a few things first 

> OpenBLAS : Your OS does not support AVX instructions. OpenBLAS is using Nehalem kernels as a fallback, which may give poorer performance.

This means your cpu does not support the AVX instruction set so it will take longer.

> /home/codelast/programme/pi/ELL/build/tutorials/vision/gettingStarted/compiled_vgg16ImageNet_pi3/vgg16ImageNet.o for pi3
Killed

And Here it looks like the kernel is killing the build process for using too many resources.

codelast, 2017-07-27:
@DarrenRainey Seems that you were right. I switched to darknet and the compilation successfully finished.
So it must be that compiling the CNTK model using too much resource on my old PC :(

codelast, 2017-07-27:
@lisaong As you suggested, I just tried darknet and successfully compiled it on my Ubuntu PC.
Thank you so much.

closed 2017-08-07 by codelast