microsoft/ELL #63: compile command not known on macOS Sierra

h4gen, 2017-07-26:
Hi everybody!

I was able to go through all examples without problems. But when I tried the examples from ELL/examples/compiled_code, I neither had a Release directory nor a compile command. Do I have to install any further dependencies to make it work or do I oversee something?
Thanks for your help! Great work you guys are doing here!

lisaong, 2017-07-26:
Hi,

Thanks for bringing this up and glad the tutorial examples worked for you!  We've been focused more on the tutorials because they help demonstrate a real use case.  

As a result, the examples/compiled_code stuff are out of date, and probably need a revamp.

h4gen, 2017-07-26:
Thanks for the update! But how can I compile a model that is not one of the examples? Can you tell me where the entry point is to figure out how the translation from model to LLVM IR works? I don't know if I am missing something, but it seems to me that, apart from the examples, this is essentially what ELL is all about?

lisaong, 2017-07-26:
Ah, I get your question now.  

If you look at cntkDemo.py and darknetDemo.py, you'll see the models being imported from the python scripts.  Basically, the idea is you pass model files (from CNTK or darknet) you want to try to those scripts.  It should be pretty clear from the script where you can pass in your own model files, just look for these lines:

```
    # Pick the model you want to work with
    helper = mh.ModelHelper(sys.argv, "darknetReference", ["darknet.cfg", "darknet.weights"], "darknetImageNetLabels.txt")

    # Import the model
    model = get_ell_predictor(helper)
```

At this time we've only verified darknetReference and vgg16, but you can always pass in a different model from either framework.  

Keep in mind though that some of the other models (e.g. tiny-yolo) are huge, which makes it harder to compile and link against.  We're aware of this issue with off the shelf models, and are working to resolve it.  For now, I recommend choosing models that are similar in size with darknet.weights for best results.

h4gen, 2017-07-26:
Thanks for your kind Answer. 
Sorry maybe the explanation of my Problem was to short. Just using the script does not help me. I am in particular interested in the build process, as I want to deploy a model on an ARM Cortex M4. For this I have to understand the build process and the cross compilation process. Right now, as it seems to me, the really interesting part, namely the build process, is hidden within huge makefiles. That is why I asked for the compile command, which does not work, but it seemed as it should work. I want to understand how I can compile a small model as in ELL/examples/compiled_code for a different target than the host or an raspberry3. I assume that this is possible, as you have an example for ARM Cortex M0 and state that models can be deployed to other architectures. I just don't see where this happens, or where I can entry to do it on myself.

Thanks for your effort.

lisaong, 2017-07-26:
(Updated the title, thanks for the feedback)

The compile commands are documented in [compilingAdvanced](https://github.com/Microsoft/ELL/blob/master/tutorials/vision/gettingStarted/compilingAdvanced.md). The commands are written for a Windows host machine, but you should be able to adapt for a Linux or Mac host machine.

h4gen, 2017-07-26:
Thank you for your answer. This article has the same Problem as the other.
It uses the compile command. But this command does not work. And I have no
chance of fixing this error because it is explained nowhere. Is it a
dependency? Is it defined during the make of ELL? WHAT is the compile
command? Also there is no Release folder. So I think that something is
going wrong when building ELL?

h4gen, 2017-07-26:
Okay, I found it. It seems, as that the compile executable is not located in the ELL/build/bin/Release, but in the ELL/build/bin folder. Now it makes sense, that you have to add this directory to the PATH before you can use it.

That answered my question.

lisaong, 2017-07-26:
Correct, for non-Windows (i.e. Linux and Mac), the compile executable is in the build/bin folder.  

For Windows, you add the "Release" or "Debug" subfolders because of the way Visual Studio generates the configurations.

vrbala, 2017-07-27:
@h4gen Hi, I am also interested in using ELL library in Cortex series micro controllers. I am barely a starter and would like to know more about this and looking forward for options to  collaborate. 
Could you provide any pointers on how to start on this please? Currently I am spending most of my time on learning the Cortex ecosystem.

Many thanks!

vrbala, 2017-07-27:
Thanks @h4gen! This is a very good starter for me.

closed 2017-08-01 by h4gen