microsoft/ELL #67: _darknetReference.so generated on RPi, but failed to run compiledDarknetDemo.py

codelast, 2017-07-27:
I was able to finish all the steps in tutorial and finally build a _darknetReference.so Python module on RPi 3, but when I run **compiledDarknetDemo.py** under the conda env, it gives the following error message:

> (py34)[root@alarmpi compiled_darknetReference_pi3]# python compiledDarknetDemo.py ~/cat.jpg                                                                  
> Traceback (most recent call last):                                                                                                                           
>   File "/root/raspberry-pi/ai/ell-related/compiled_darknetReference_pi3/darknetReference.py", line 14, in swig_import_helper                                 
>     return importlib.import_module(mname)                                                                                                                    
>   File "/root/.miniconda3/envs/py34/lib/python3.4/importlib/__init__.py", line 109, in import_module                                                         
>     return _bootstrap._gcd_import(name[level:], package, level)                                                                                              
>   File "<frozen importlib._bootstrap>", line 2254, in _gcd_import                                                                                            
>   File "<frozen importlib._bootstrap>", line 2237, in _find_and_load                                                                                         
>   File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked                                                                                
>   File "<frozen importlib._bootstrap>", line 1191, in _load_unlocked                                                                                         
>   File "<frozen importlib._bootstrap>", line 1161, in _load_backward_compatible                                                                              
>   File "<frozen importlib._bootstrap>", line 539, in _check_name_wrapper                                                                                     
>   File "<frozen importlib._bootstrap>", line 1715, in load_module
>   File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
> ImportError: build/_darknetReference.so: undefined symbol: cblas_sgemm                              
>                                                                
> During handling of the above exception, another exception occurred:
>                                                                               
> Traceback (most recent call last):                                            
>   File "compiledDarknetDemo.py", line 11, in <module>                         
>     import darknetReference as model                                          
>   File "/root/raspberry-pi/ai/ell-related/compiled_darknetReference_pi3/darknetReference.py", line 17, in <module>
>     _darknetReference = swig_import_helper()               
>   File "/root/raspberry-pi/ai/ell-related/compiled_darknetReference_pi3/darknetReference.py", line 16, in swig_import_helper
>     return importlib.import_module('_darknetReference')            
>   File "/root/.miniconda3/envs/py34/lib/python3.4/importlib/__init__.py", line 109, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)               
> ImportError: build/_darknetReference.so: undefined symbol: cblas_sgemm

And following are the output messages of "cmake .." command on RPi 3 to generate the _darknetReference.so Python module:

> (py34)[root@alarmpi build]# cmake ..               
> -- The C compiler identification is GNU 7.1.1      
> -- The CXX compiler identification is GNU 7.1.1    
> -- Check for working C compiler: /usr/bin/cc       
> -- Check for working C compiler: /usr/bin/cc -- works                                                 
> -- Detecting C compiler ABI info                   
> -- Detecting C compiler ABI info - done            
> -- Detecting C compile features                    
> -- Detecting C compile features - done             
> -- Check for working CXX compiler: /usr/bin/c++    
> -- Check for working CXX compiler: /usr/bin/c++ -- works                                              
> -- Detecting CXX compiler ABI info                 
> -- Detecting CXX compiler ABI info - done          
> -- Detecting CXX compile features                  
> -- Detecting CXX compile features - done           
> -- Looking for pthread.h 
> -- Looking for pthread.h - found                   
> -- Looking for pthread_create                      
> -- Looking for pthread_create - not found          
> -- Looking for pthread_create in pthreads          
> -- Looking for pthread_create in pthreads - not found                                                 
> -- Looking for pthread_create in pthread           
> -- Looking for pthread_create in pthread - found   
> -- Found Threads: TRUE   
> -- Blas libraries: /usr/lib/libblas.so             
> -- Blas linker flags:    
> -- Blas include directories:                       
> -- Using BLAS include path: /usr/include           
> -- Using BLAS library: /usr/lib/libblas.so         
> -- Using BLAS DLLs:      
> -- Found PythonInterp: /root/.miniconda3/envs/py34/bin/python3.4 (found suitable version "3.4.3", minimum required is "3.4") 
> -- Found PythonLibs: /root/.miniconda3/envs/py34/lib/libpython3.4m.so (found suitable version "3.4.3", minimum required is "3.4") 
> -- Configuring done      
> -- Generating done       
> -- Build files have been written to: /root/raspberry-pi/ai/ell-related/compiled_darknetReference_pi3/build

Could anyone tell me what am I missing?
Thanks a lot.

DarrenRainey, 2017-07-27:
What version of python are you using ? . You might need to update to 3.6 as it looks like your running 3.4.3

lisaong, 2017-07-27:
Hi,

python 3.4.3 should be ok on the Raspberry Pi3, because that's what the Miniconda environment currently supports.

The "cblas_dgemm" undefined symbol is potentially a mismatch of blas libs on your pi system.  For example, another version of blas taking precedence over openblas and getting picked up.

Take a look at these issues to see if they help: https://github.com/Microsoft/ELL/issues?utf8=%E2%9C%93&q=is%3Aissue%20cblas_dgemm

In particular, check whether libblas.so has that symbol defined.  If not, and if you did run "sudo apt-get install libopenblas-dev", then check whether there are other blas .so that have the correct symbol.

codelast, 2017-08-01:
Thank you for your reply.
These days I tried several methods but none of them works for me :(
I don't have atlas installed on my RPi, and I check the blas library .so file printed by the cmake command, it does contains the cblas_sgemm function, e.g. 

> nm -D xxx.so | grep cblas_sgemm
(The output it NOT empty, e.g. "0005f230 T cblas_sgemm")

So I havn't figured out this problem yet, if anyone knows anything that might help solving this problem, please post a comment here, thank you in advance.

lisaong, 2017-08-01:
Thanks for verifying that you have the right lib. The only other thing I can think of is that your conda environment is somehow not finding the .so at runtime.  

Does your LD_LIBRARY_PATH contain /usr/lib?

>File "/root/.miniconda3/envs/py34/lib/python3.4/importlib/init.py", line 109, in import_module
> return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: build/_darknetReference.so: undefined symbol: cblas_sgemm

codelast, 2017-08-05:
@lisaong I figure it out finally :)
Here is my solution(all following steps are executed on RPi 3):
* Uninstall the blas & cblas library installed by "pacman -S blas cblas"(I'm using Arch Linux ARM on RPi 3, command **pacman -S xxx** is just like the **apt-get install xxx** on Raspbian)
* Download the OpenBlas source code & compile it from souce, then install it to /usr/lib/openblas
* Before "cmake ..", set the **LD_LIBRARY_PATH** to let the openblas lib directory be found by cmake:

> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/openblas/lib
* make a little change to the **OpenBLASSetup.cmake** file, add path **/usr/lib/openblas/include/** to the **set(BLAS_INCLUDE_SEARCH_PATHS** section, that is, the section will looks like:

>     set(BLAS_INCLUDE_SEARCH_PATHS
>         /System/Library/Frameworks/Accelerate.framework/Versions/Current/Frameworks/vecLib.framework/Versions/Current/Headers/
>         /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Accelerate.framework/Frameworks/vecLib.framework/Headers/
>         /usr/include
>         /usr/local/include
>         /usr/lib/openblas/include
>     )

* After that, the blas-related messages in cmake output looks like:

> -- Blas libraries: /usr/lib/libblas.so
> -- Blas linker flags: 
> -- Blas include directories: 
> -- Using BLAS include path: /usr/lib/openblas/include
> -- Using BLAS library: /usr/lib/openblas/lib/libopenblas.so

* Then make the Python module _darknetReference.so, and test it, it works!

Finally I would like to say thank you to all of your ELL developers, you gave me a lot of help in the process of testing it, and it's a quite good project.

lisaong, 2017-08-05:
Great to see you found a fix!

closed 2017-08-06 by codelast