microsoft/ELL #68: [ERROR] python darknetDemo.py

hustzxd, 2017-07-28:
`====================================================================
ELL module is not loading
It is possible that you need to add LibOpenBLAS to your system path (See Install-*.md) from root of this repo
Traceback (most recent call last):
  File "./../../../../build/interfaces/python/ELL.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/zxd/miniconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 648, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 560, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
ImportError: /home/zxd/miniconda3/envs/py36/lib/python3.6/site-packages/../../libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./../../../../build/interfaces/python/_ELL.so)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "darknetDemo.py", line 7, in <module>
    import darknet_to_ell
  File "./../../../../build/tools/importers/darknet/darknet_to_ell.py", line 17, in <module>
    import ELL
  File "./../../../../build/interfaces/python/ELL.py", line 17, in <module>
    _ELL = swig_import_helper()
  File "./../../../../build/interfaces/python/ELL.py", line 16, in swig_import_helper
    return importlib.import_module('_ELL')
  File "/home/zxd/miniconda3/envs/py36/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/zxd/miniconda3/envs/py36/lib/python3.6/site-packages/../../libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./../../../../build/interfaces/python/_ELL.so)
`

DarrenRainey, 2017-07-28:
Looks like its not finding OpenBLAS from this line

> ELL module is not loading It is possible that you need to add LibOpenBLAS to your system path (See Install-*.md) from root of this repo 

Maybe try  apt install libopenblas-dev or if you already have it installed add it to your $PATH

lisaong, 2017-08-01:
This error means that _ELL.so was compiled with a different version of libstdc++.so.6 from what miniconda binaries are built against. 

```
ImportError: /home/zxd/miniconda3/envs/py36/lib/python3.6/site-packages/../../libstdc++.so.6: version CXXABI_1.3.9' not found (required by ./../../../../build/interfaces/python/_ELL.so)
```

There are a couple of ways to fix this, but the easiest one in my experience is to:
1. Make sure you have the library somewhere (you should have it on the system since you built _ELL.so)
2. export LD_LIBRARY_PATH=path_to_that_libstdc++.so:$LD_LIBRARY_PATH

hustzxd, 2017-08-07:
Thank you!   :)
`export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6`

closed 2017-08-07 by hustzxd