microsoft/ELL #70: ** On entry to SGEMM  parameter number  8 had an illegal value [Pi Zero]

pnvphuong, 2017-07-30:
I am trying to deploy the darknet tutorial on Pi Zero. The model has been successfully compiled on Ubuntu.
On Pi Zero, I recompiled directly on the main environment instead of Miniconda. Because `Miniconda3-latest-Linux-armv7l.sh` does not support Pi Zero (armv6l). Meanwhile, `Miniconda-3.5.5-Linux-armv6l.sh` supports Pi Zero but can't install python 3 environment, which is required by the tutorials.

The compiling process does not raise any troubles
```
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Blas libraries: /usr/lib/libopenblas.so
-- Blas linker flags:
-- Blas include directories:
-- Using BLAS include path: /usr/include
-- Using BLAS library: /usr/lib/libopenblas.so
-- Using BLAS DLLs:
-- Found PythonInterp: /usr/bin/python3.4 (found suitable version "3.4.2", minimum required is "3.4")
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython3.4m.so (found suitable version "3.4.2", minimum required is "3.4")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/ELL_demo/build
```

However, when I run the demo `python compiledDarknetDemo.py`, there are errors
```
** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value
 ** On entry to SGEMM  parameter number  8 had an illegal value

(frame:1241): Gtk-WARNING **: cannot open display:
```

I have checked the system only has `libopenblas-dev`, no `libatlas-dev`

Would you help me with this issue?
Thank you

lisaong, 2017-07-31:
This is a known issue that we are still investigating. No ETA yet at this time but we are open to suggestions from anyone.

pnvphuong, 2017-07-31:
Thanks a lot for your reply, lisaong. I hope we can find a solution soon for Pi Zero. 
I just wonder if the errors come from ELL compiled code itself or from the tutorials, e.g. using `opencv`.

h4gen, 2017-08-01:
@pnvphuong What happens, when you just comment out the whole opencv part in the python script? Looks pretty much like an opencv issue to me. I would be surprised, if the ELL compiled code calls any gtk functions itself.

lisaong, 2018-01-09:
New instructions for Raspberry Pi Zero are now published with release v2.1.5.

closed 2018-01-09 by lisaong