microsoft/ELL #73: Error on building ELL on Ubuntu 16.04.2

Sneha369, 2017-08-03:
Hello,

I am trying build ELL on my Ubuntu 16.04.2  32 bit

My build is failing at
[ 44%] Building CXX object libraries/nodes/CMakeFiles/nodes.dir/src/BinaryConvolutionalLayerNode.cpp.o

Here is the full ouput of **make** 

Scanning dependencies of target documentation
[  0%] Built target documentation
Scanning dependencies of target utilities
[  0%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Archiver.cpp.o
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CommandLineParser.cpp.o
[  1%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CompressedIntegerList.cpp.o
/home/rk/ELL/libraries/utilities/src/CompressedIntegerList.cpp: In member function ‘void ell::utilities::CompressedIntegerList::Reset()’:
/home/rk/ELL/libraries/utilities/src/CompressedIntegerList.cpp:141:17: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         _last = UINT64_MAX;
                 ^~~~~~~~~~
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ConformingVector.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/CStringParser.cpp.o
[  2%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Files.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Format.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IArchivable.cpp.o
[  3%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IndentedTextWriter.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerList.cpp.o
[  4%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/IntegerStack.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/JsonArchiver.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchive.cpp.o
[  6%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/ObjectArchiver.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/OutputStreamImpostor.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/PPMImageParser.cpp.o
[  7%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/RandomEngines.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Tokenizer.cpp.o
[  8%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/TypeName.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/UniqueId.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/Variant.cpp.o
[  9%] Building CXX object libraries/utilities/CMakeFiles/utilities.dir/src/XmlArchiver.cpp.o
[ 11%] Linking CXX static library libutilities.a
[ 11%] Built target utilities
Scanning dependencies of target math
[ 11%] Building CXX object libraries/math/CMakeFiles/math.dir/src/BlasWrapper.cpp.o
[ 11%] Linking CXX static library libmath.a
[ 11%] Built target math
Scanning dependencies of target data
[ 11%] Building CXX object libraries/data/CMakeFiles/data.dir/src/Dataset.cpp.o
[ 11%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVector.cpp.o
[ 12%] Building CXX object libraries/data/CMakeFiles/data.dir/src/DataVectorOperations.cpp.o
[ 12%] Building CXX object libraries/data/CMakeFiles/data.dir/src/GeneralizedSparseParsingIterator.cpp.o
[ 12%] Building CXX object libraries/data/CMakeFiles/data.dir/src/SequentialLineIterator.cpp.o
[ 13%] Building CXX object libraries/data/CMakeFiles/data.dir/src/TextLine.cpp.o
[ 13%] Building CXX object libraries/data/CMakeFiles/data.dir/src/WeightLabel.cpp.o
[ 14%] Linking CXX static library libdata.a
[ 14%] Built target data
Scanning dependencies of target emitters
[ 16%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/EmitterTypes.cpp.o
[ 16%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRAssemblyWriter.cpp.o
[ 17%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRBlockRegion.cpp.o
[ 17%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRDiagnosticHandler.cpp.o
[ 17%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IREmitter.cpp.o
[ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRExecutionEngine.cpp.o
[ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRFunctionEmitter.cpp.o
[ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRHeaderWriter.cpp.o
[ 19%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRIfEmitter.cpp.o
[ 19%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRLoader.cpp.o
[ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRLoopEmitter.cpp.o
[ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRMetadata.cpp.o
[ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRModuleEmitter.cpp.o
[ 22%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IROptimizer.cpp.o
[ 22%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRRuntime.cpp.o
[... 373 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rk/ELL/build

**### I am using gcc version 6.3 and cmake version 3.8.2** 

for the version of llvm (**apt-cache show llvm-dev** )
Package: llvm-dev
Priority: optional
Section: universe/devel
Installed-Size: 27
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: LLVM Packaging Team <pkg-llvm-team@lists.alioth.debian.org>
Architecture: i386
Source: llvm-defaults (0.33ubuntu3)
Version: 1:3.8-33ubuntu3
Depends: llvm-runtime (= 1:3.8-33ubuntu3), llvm-3.8-dev (>= 3.8~), llvm
Filename: pool/universe/l/llvm-defaults/llvm-dev_3.8-33ubuntu3_i386.deb

even though I installed llvm3.9-dev and also tried configuring using update-alternatives 
it is showing version 3.8

Could anyone tell me how to solve both of these issues?

Thanks

lisaong, 2017-08-04:
Thanks for reporting these.  We have been mostly running on Ubuntu 64 bit, and will need to investigate these errors that look related to 32-bit.  

Meanwhile, if it is feasible for you, is it possible for you to use 64 bit?

Sneha369, 2017-08-05:
okay. I will try it out on the 64 bit. Thanks for the reply :)

Sneha369, 2017-08-07:
Hi,
I have tried it on 64 bit , and everything worked!! Thank you.

closed 2017-08-10 by Sneha369