microsoft/ELL #74: OpenBLAS tests failed on Windows

lmiroslaw, 2017-08-03:
Dear Team,

I compiled ELL and Python bindings successfully. I also updated PATH to indicate the location of OpenBLAS. Unfortunately not all the tests have passed.

> 
> 84% tests passed, 5 tests failed out of 32
> Total Test time (real) = 544.51 sec
> 
> The following tests FAILED:
>           6 - math_test (Failed)
>           8 - model_compiler_test (Failed)
>           9 - nodes_test (Failed)
>          10 - predictors_test (Failed)
>          14 - darknet_importer_test (ILLEGAL)

Presumambly the first error is related to libopenblas:
`Unhandled exception at 0x000000006DA4FA74 (libopenblas.dll) in math_test.exe: 0xC000001D: Illegal Instruction.`

These are some more detailed errors (added -VV to ctest):

> test 14
>       Start 14: darknet_importer_test
> 
> 14: Test command: C:\Users\lumirosl\AppData\Local\Programs\Python\Python36\python.exe "-m" "unittest" "darknet_to_ell_importer_test.py"
> 14: Test timeout computed to be: 9.99988e+006
> 14/32 Test #14: darknet_importer_test ............***Failed    5.67 sec
> 
> 6: Test timeout computed to be: 9.99988e+006
>  6/32 Test  #6: math_test ........................***Failed    5.46 sec
> 
> 8: Test command: C:\workspace\ELL\build\libraries\model\Release\model_compiler_test.exe
> 8: Test timeout computed to be: 9.99988e+006
>  8/32 Test  #8: model_compiler_test ..............***Failed    4.94 sec

lisaong, 2017-08-04:
Hi,

This sounds like an incompatibility with the open blas library and your processor.  Are you running an Intel CPU, and if yes, what chipset is it?

You can find the logic that's used to infer the processor from here: https://github.com/Microsoft/ELL/blob/master/CMake/OpenBLASSetup.cmake, around line 63.

CheekyMonkey, 2017-08-05:
Hi, I observe the same issue:

82% tests passed, 6 tests failed out of 33

Total Test time (real) = 113.06 sec

The following tests FAILED:
          6 - math_test (Failed)
          8 - model_compiler_test (Failed)
          9 - nodes_test (ILLEGAL)
         10 - predictors_test (ILLEGAL)
         14 - darknet_importer_test (ILLEGAL)
         32 - ell_module_test (Failed)
Errors while running CTest

My CPU is "Mobile Intel Core i5 520M (Arrandale)"

My registry value for the following key
\HKEY_LOCAL_MACHINE\HARDWARE\DESCRIPTION\System\CentralProcessor\0

is "Intel64 Family 6 Model 37 Stepping 5"

The CPU types shown in OpenBLASSetup.cmake are:

        # Haswell: Family 6, model 60, 63, 69
        # Broadwell: Family 6, Model 70, 79 (compatible with Haswell)
        # Kobylake: Family 6, Model 78 (compatible with Haswell)
        # Sandybridge: Family 6, model 42, 45
        # Ivybridge: Family 6, model 58 (compatible with Sandybridge)
        # Skylake: Family 6, model 42

Is there a processor-specific version of OpenBLAS for my particular CPU?

[image]

[image]

lisaong, 2017-08-05:
Thanks for the details.  Since your CPU is not one of those supported by our Nuget package (which targets the more common configurations), you can try building OpenBlas for your CPU target and replacing the dlls in your ELL\external\OpenBLASLibs.0.2.19.3\build\native\x64 folder.

Instructions on building OpenBlas are here:
https://github.com/xianyi/openblas/releases

CheekyMonkey, 2017-08-07:
Thanks for the guidance

That appears to have made a difference in my case

After running through the compile steps again, test outcomes are now as follows:

(py36) G:\DeepLearning\Microsoft\ELL\build>ctest --build-config Release
Test project G:/DeepLearning/Microsoft/ELL/build
      Start  1: common_test
 1/33 Test  #1: common_test ......................   Passed    0.97 sec
      Start  2: data_test
 2/33 Test  #2: data_test ........................   Passed    0.02 sec
      Start  3: emitters_test
 3/33 Test  #3: emitters_test ....................   Passed    0.45 sec
      Start  4: evaluators_test
 4/33 Test  #4: evaluators_test ..................   Passed    0.01 sec
      Start  5: functions_test
 5/33 Test  #5: functions_test ...................   Passed    0.02 sec
      Start  6: math_test
 6/33 Test  #6: math_test ........................   Passed    0.08 sec
      Start  7: model_test
 7/33 Test  #7: model_test .......................   Passed    0.29 sec
      Start  8: model_compiler_test
 8/33 Test  #8: model_compiler_test ..............   Passed    2.75 sec
      Start  9: nodes_test
 9/33 Test  #9: nodes_test .......................   Passed    0.07 sec
      Start 10: predictors_test
10/33 Test #10: predictors_test ..................   Passed    0.03 sec
      Start 11: trainers_test
11/33 Test #11: trainers_test ....................   Passed    0.02 sec
      Start 12: utilities_test
12/33 Test #12: utilities_test ...................   Passed    0.88 sec
      Start 13: cntk_importer_test
13/33 Test #13: cntk_importer_test ...............   Passed    8.72 sec
      Start 14: darknet_importer_test
14/33 Test #14: darknet_importer_test ............   Passed    0.78 sec
      Start 15: linearTrainer_test_0
15/33 Test #15: linearTrainer_test_0 .............   Passed    0.10 sec
      Start 16: linearTrainer_test_1
16/33 Test #16: linearTrainer_test_1 .............   Passed    0.11 sec
      Start 17: linearTrainer_test_2
17/33 Test #17: linearTrainer_test_2 .............   Passed    0.03 sec
      Start 18: linearTrainer_test_3
18/33 Test #18: linearTrainer_test_3 .............   Passed    0.03 sec
      Start 19: linearTrainer_test_4
19/33 Test #19: linearTrainer_test_4 .............   Passed    0.03 sec
      Start 20: linearTrainer_test_5
20/33 Test #20: linearTrainer_test_5 .............   Passed    0.03 sec
      Start 21: linearTrainer_test_6
21/33 Test #21: linearTrainer_test_6 .............   Passed    0.05 sec
      Start 22: linearTrainer_test_7
22/33 Test #22: linearTrainer_test_7 .............   Passed    0.03 sec
      Start 23: linearTrainer_test_8
23/33 Test #23: linearTrainer_test_8 .............   Passed    0.04 sec
      Start 24: linearTrainer_test_9
24/33 Test #24: linearTrainer_test_9 .............   Passed    0.03 sec
      Start 25: linearTrainer_test_10
25/33 Test #25: linearTrainer_test_10 ............   Passed    0.03 sec
      Start 26: forestTrainer_test
26/33 Test #26: forestTrainer_test ...............   Passed    0.11 sec
      Start 27: sweepingSGDTrainer_test
27/33 Test #27: sweepingSGDTrainer_test ..........   Passed    0.11 sec
      Start 28: apply_test
28/33 Test #28: apply_test .......................   Passed    0.04 sec
      Start 29: compile_test1
29/33 Test #29: compile_test1 ....................   Passed    0.09 sec
      Start 30: compile_test2
30/33 Test #30: compile_test2 ....................   Passed    0.07 sec
      Start 31: ell-python-interface-test
31/33 Test #31: ell-python-interface-test ........   Passed    0.71 sec
      Start 32: ell_module_test
32/33 Test #32: ell_module_test ..................***Failed    0.29 sec
      Start 33: ELL_step10_python_test
33/33 Test #33: ELL_step10_python_test ...........   Passed    0.09 sec

97% tests passed, 1 tests failed out of 33

Total Test time (real) =  18.92 sec

The following tests FAILED:
         32 - ell_module_test (Failed)
Errors while running CTest

The test that fails, shows the following in the test logs:

32/33 Testing: ell_module_test
32/33 Test: ell_module_test
Command: "C:/Program Files (x86)/nodejs/node.exe" "TestCompiler.js"
Directory: G:/DeepLearning/Microsoft/ELL/interfaces/javascript/ell_module/test/js
"ell_module_test" start time: Aug 07 04:50 GMT Summer Time
Output:
----------------------------------------------------------
G:\DeepLearning\Microsoft\ELL\interfaces\javascript\ell_module\test\js\TestCompiler.js:3
let path = require('path');
^^^
SyntaxError: Unexpected strict mode reserved word
    at exports.runInThisContext (vm.js:73:16)
    at Module._compile (module.js:443:25)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3
<end of output>
Test time =   0.29 sec
----------------------------------------------------------
Test Failed.
"ell_module_test" end time: Aug 07 04:50 GMT Summer Time
"ell_module_test" time elapsed: 00:00:00
----------------------------------------------------------

Are you able to share any guidance on how to overcome that test failure?

lmiroslaw, 2017-08-16:
I have an Ivybridge and had to recompile OpenBLAS.
I also replaced the libs in the mentioned directory with the new ones:

1049089    99328 Aug  3 10:44 libgcc_s_seh-1.dll
1049089  1220096 Aug  3 10:44 libgfortran-3.dll
1049089 17663597 Aug  3 10:44 libopenblas.dll
1049089   346112 Aug  3 10:44 libquadmath-0.dll

Now, the passing rate is a bit greater: 

88% tests passed, 4 tests failed out of 32

Total Test time (real) =  37.98 sec

The following tests FAILED:
          6 - math_test (Failed)
          8 - model_compiler_test (Failed)
          9 - nodes_test (Failed)
         10 - predictors_test (Failed)
Errors while running CTest