microsoft/ELL #75: Windows 10 64 build errors ..

ack72tdp, 2017-08-04:
Any ideas on what I missed ..? 

I had all installed properly ..
Full Anaconda
nuget
openBLAS
C:\Program Files (x86)\Microsoft Visual Studio\2017
and CMAKE

and all paths are set right .. 

(py36) c:\MSELL\ELL\build>cmake -G "Visual Studio 15 2017 Win64" -DPROCESSOR_HINT=haswell ..
CMake Error at CMakeLists.txt:7 (project):
  Failed to run MSBuild command:

    MSBuild.exe

  to get the value of VCTargetsPath:

-- Configuring incomplete, errors occurred!
See also "C:/MSELL/ELL/build/CMakeFiles/CMakeOutput.log".

(py36) c:\MSELL\ELL\build>cmake --build . --config Release
The system cannot find the file specified
CMake Error: Generator: execution of make failed. Make command was: "MSBuild.exe" "ALL_BUILD.vcxproj" "/p:Configuration=Release" "/p:VisualStudioVersion=15.0"

ack72tdp, 2017-08-04:
output of C:/MSELL/ELL/build/CMakeFiles/CMakeOutput.log". just says ..  
The system is: Windows - 10.0.10586 - AMD64

ack72tdp, 2017-08-04:
MSBuild appear to be in path

(py36) c:\MSELL\ELL\build>MSbuild
Microsoft (R) Build Engine version 15.1.1012.6693
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1003: Specify a project or solution file. The current working directory does not contain a project or solution file.

ack72tdp, 2017-08-04:
never mind .. missed ... Desktop Development with C++

closed 2017-08-04 by ack72tdp

y-vyrovoy, 2017-11-16:
so what was the solution?

kernhanda, 2017-11-16:
@y-vyrovoy, looks like he forgot to select "Desktop Development with C++" when he was running the VS installer.

Weiney, 2018-07-16:
I have the same problem today,to solve this
add ms path to the Environmental path
like  D:\Microsoft Visual Studio\MSBuild\15.0\Bin

baluoteliz, 2019-12-06:
ye,i also encontered the same problem.
,the solution is rechoose the vs tools for your environment.
as below:
vc14 vs2015
vc15 vs 2017...