microsoft/ELL #78: will it work on esp32 ?

lovettchris, 2017-08-09:
saurabhvyas asked whether ELL will work with esp32.  The esp32 chip is cool, and I love the sparkfun board they made for it.  Unfortunately the Xtensa LX6 CPU is not one that LLVM knows how to target and ELL uses LLVM to compile the model for a given target platform.  The easiest approach would be to take the compiled LLVM IR and generate C code from it then use the Xtensa C++ compiler which knows how to vectorize for that chip.  LLVM used to be able to generate C code, but that feature was also removed, unfortunately.  So the work could be done inside ELL IRModuleEmitter, and it would be similar to LLVM::Modulde::print, only it would print C instead of LLVM IR.  This would be simpler than general C LLVM support given the limited LVM constructs used by ELL.  You could probably dig up some old LLVM C generation code from someplace and run with that.  An alternative approach might be to play with the llvm-gcc compiler which seems to support Xtensa, and it can link the LLVM bitcode files which is generated by ELL... but I'm not sure how good the llvm-gcc compiler is compared to the dedicated Xtensa compiler when it comes to taking full advantage of the Xtensa chip capabilities.

closed 2017-10-30 by lovettchris

grahamehorner, 2018-12-16:
@lovettchris IMHO the esp32 is a chip that I feel is ideal for edge processing where low power WiFi or Bluetooth is used for communication, I have personally used these a number of times in personal projects with great success. IMHO ELL would be a great on these devices and/or aurduio device maybe an ELL arduino runtime library is a possible option if ELL is not able/going to support native compilation for the ESP32 FreeRTOS ICs ? 

I would love to see this re-opened to allow more discussion on ELL and ICs that aren’t covered with the LLVM compiler generally.

grahamehorner, 2018-12-18:
FYI 
https://github.com/Sonicadvance1/llvm-xtensa