microsoft/ELL #79: ELL Benchmark

dayanandasiet, 2017-08-10:
Dear Team,

Would you please provide the Benchmark for ELL when compare to Caffe, Caffe2, Tensorflow or others. Benchmark will help to understand the performance on respected target.

butterl, 2017-08-18:
I  also interested in this ， for if we can get a clear look on the performance and use case opptimiztion , ELL would get more people in .  @lisaong  I find you are answering many issue， would you like to help  to share it :)

codelast, 2017-08-23:
I made a very, very simple comparison between ELL(Darknet model) & TensorFlow(Inception v3 model)  on doing image classification jobs, the result shows that:
* the inference speed of ELL+Darknet is much faster than TensorFlow+Inception v3
* the inference accuracy of ELL+Darknet is much worse than TensorFlow+Inception v3

The result can be found [here](https://www.codelast.com/?p=9778)(only has Chinese version, sorry)

lisaong, 2017-08-23:
Thanks @codelast for taking the time to do the comparison and posting the results!

For **accuracy** benchmarking, here are a few suggestions on what to try next:
- Since you are using photos, which is great as a test because it eliminates webcam variances, you can perform the comparison on your desktop machine.  This should save you some time.  Speed comparisons should still be done on the Raspberry Pi though.

For accuracy, in general you'd want to compare models that are equivalent.  Since ELL basically takes a pre-trained model and compiles it to run faster, the model's accuracy is dependent on how that model is trained.

For example, for darknetReference, you can compare the accuracy of the original (darknetReference model) [https://pjreddie.com/darknet/imagenet/] (search for "Darknet Reference" on the page for instructions on how to get it up and running) with ELL's compiled version.  TensorFlow doesn't readily have a darknetReference model available because it requires TensorFlow to support importing darknet models.

In other words, your comparison can be done this way:
1. Accuracy (on PC):  original darknetReference (from the pjreddie site) vs. ELL darknetReference
2. Speed (on PC):  original darknetReference (from the pjreddie site) vs. ELL darknetReference.  If you can use GPU for original darknetReference, I expect the original to be faster.
3. Speed (on Pi3):  original darknetReference (from the pjreddie site) vs. ELL darknetReference.  I'm not sure if the original supports compiling on the Pi3 though. 

Ideally, it will be nice to be able to compare ELL(Inception v3 model) with TensorFlow(Inception v3 model).  This is good feedback but not yet supported.

lovettchris, 2017-08-23:
Also, when testing a model it doesn't make sense to ask it to recognize images that it wasn't trained on.  So your test set should be limited to these categories [ImageNet 1k categories ](http://image-net.org/challenges/LSVRC/2012/browse-synsets) and notice these do not include any flowers or kids toys or apples.  So if you remove those from your test set you have 7 and ELL got 5 out of 7.  It is unfortunate it didn't get the bicycle, but it was trained on [mountain bikes](http://image-net.org/synset?wnid=n03792782) only.

closed 2017-10-02 by lovettchris