microsoft/ELL #86: Cannot build on Manjaro

stephansgit, 2017-08-28:
Dear all,

I'm trying to build on Manjaro:
`cat /proc/version` gives
`Linux version 4.9.44-1-MANJARO (builduser@manjaro) (gcc version 7.1.1 20170630 (GCC) ) #1 SMP PREEMPT Thu Aug 17 08:23:52 UTC 2017`

`cmake ..` runs just fine,

but `make` produces the following error:

    [  0%] Built target documentation  
    [ 10%] Built target utilities  
    [ 12%] Built target math
    [ 15%] Built target data
    Scanning dependencies of target emitters
    [ 15%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/EmitterTypes.cpp.o
    [ 15%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRAssemblyWriter.cpp.o
    [ 16%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRBlockRegion.cpp.o
    [ 16%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRDiagnosticHandler.cpp.o
    [ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IREmitter.cpp.o
    [ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRExecutionEngine.cpp.o
    [ 18%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRFunctionEmitter.cpp.o
    [ 19%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRHeaderWriter.cpp.o
    [ 19%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRIfEmitter.cpp.o
    [ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRLoader.cpp.o
    [ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRLoopEmitter.cpp.o
    [ 20%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRMetadata.cpp.o
    [ 21%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRModuleEmitter.cpp.o
    [ 21%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IROptimizer.cpp.o
    [ 21%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRRuntime.cpp.o
    [ 22%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/IRSwigInterfaceWriter.cpp.o
    [ 22%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/ModuleEmitter.cpp.o
    [ 24%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/TargetDevice.cpp.o
    [ 24%] Building CXX object libraries/emitters/CMakeFiles/emitters.dir/src/Variable.cpp.o
    [ 24%] Linking CXX static library libemitters.a
    [ 24%] Built target emitters
    [ 25%] Built target evaluators
    [ 28%] Built target functions 
    Scanning dependencies of target model
    [ 28%] Building CXX object libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o
    In file included from /home/stephan/ELL/libraries/model/include/Model.h:13:0,
                 from /home/stephan/ELL/libraries/model/include/IRModelProfiler.h:11,
                 from /home/stephan/ELL/libraries/model/include/CompilableNode.h:11,
                 from /home/stephan/ELL/libraries/model/src/CompilableNode.cpp:9:
    /home/stephan/ELL/libraries/model/include/PortElements.h: In member function ‘const   ell::model::OutputPort<ValueType>* ell::model::PortElements<ValueType>::ReferencedPort() const’:
    /home/stephan/ELL/libraries/model/include/PortElements.h:434:135: error: ‘ell::model::PortElementBase’ is not a base of ‘const ell::model::PortElements<ValueType>’
         const OutputPort<ValueType>* ReferencedPort() const { return static_cast<const OutputPort<ValueType>*>(this->PortElementBase::ReferencedPort()); }
                                                                                                                                       ^~~~~~~~~~~~~~
    make[2]: *** [libraries/model/CMakeFiles/model.dir/build.make:63: libraries/model/CMakeFiles/model.dir/src/CompilableNode.cpp.o] Error 1
    make[1]: *** [CMakeFiles/Makefile2:843: libraries/model/CMakeFiles/model.dir/all] Error 2
    make: *** [Makefile:95: all] Error 2

Is this related to #14 and the gcc version? I use `gcc 7.1.1`. 
However, running `make CC=gcc-5.4.0-1` didn't help either (well, I read that's how one can specify a certain gcc version for make, but I'm absolute noob).

I'd be thankful for any assistance on this!

kernhanda, 2017-10-12:
Fixed with 2d36e7d21c184f87947cda2fd18db2fc743de683.

closed 2017-10-12 by kernhanda

stephansgit, 2017-10-15:
Thanks! Works like a charm...