microsoft/ELL #97: Cannot open include file: 'Python.h': No such file or directory

amitmc19, 2017-09-23:
Hello,

I was able to get everything working for building the ELL on windows other than the Python language bindings for ELL. 
 I get a compiler error when I build the python Language bindings for ELL. (cmake --build . --target _ELL_python --config Release )
I am not familiar with cmake much, but I poked around to debug this.  I see that the CMakeCache.txt located at ELL-master\build has the below contents

//Path to a library.
PYTHON_DEBUG_LIBRARY:FILEPATH=PYTHON_DEBUG_LIBRARY-NOTFOUND

//Path to a program.
PYTHON_EXECUTABLE:FILEPATH=C:/Users/Amit/Anaconda2/python.exe

//Path to a file.
PYTHON_INCLUDE_DIR:PATH=C:/Users/Amit/Anaconda2/include

//Path to a library.
PYTHON_LIBRARY:FILEPATH=C:/Users/Amit/Anaconda2/libs/python27.lib
It does include the Python.h. But I am not sure why it is  not getting included in the build.

Below is the error, can you please provide some inputs for this?

ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\HostX86\x64\CL.exe /c /I"C:\Softwares\DeepLearning\ELL-master\external\LLVMNativeWindowsLibs.x64.3.9.1\b
  uild\native\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\common\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\common\include" /I"C:\Softwares\DeepLearning
  \ELL-master\interfaces\..\libraries\data\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\evaluators\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\librarie
  s\functions\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\model\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\nodes\include" /I"C:\Softwares\D
  eepLearning\ELL-master\interfaces\..\libraries\predictors\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\predictors\neural\include" /I"C:\Softwares\DeepLearning\ELL-maste
  r\interfaces\..\libraries\trainers\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\..\libraries\utilities\include" /I"C:\Softwares\DeepLearning\ELL-master\interfaces\python" /I"C:\Soft
  wares\DeepLearning\ELL-master\libraries\common\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\data\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\math\include" /I"C:\Softwa
  res\DeepLearning\ELL-master\external\OpenBLASLibs.0.2.19.3\build\native\x64\haswell\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\utilities\include" /I"C:\Softwares\DeepLearning\ELL-m
  aster\libraries\functions\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\model\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\emitters\include" /I"C:\Softwares\DeepLearning
  \ELL-master\libraries\nodes\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\predictors\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\predictors\neural\include" /I"C:\Softwa
  res\DeepLearning\ELL-master\libraries\evaluators\include" /I"C:\Softwares\DeepLearning\ELL-master\libraries\trainers\include" /nologo /W3 /WX- /diagnostics:classic /MP /O2 /Ob2 /D WIN32 /D _WINDOW
  S /D NDEBUG /D UNICODE /D _SCL_SECURE_NO_WARNINGS /D SWIG_PYTHON_INTERPRETER_NO_DEBUG /D USE_BLAS=1 /D /D_SCL_SECURE_NO_WARNINGS /D "CMAKE_INTDIR=\"Release\"" /D _ELL_python_EXPORTS /D _WINDLL /D
  _MBCS /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Zc:rvalueCast /GR /Fo"_ELL_python.dir\Release\\" /Fd"_ELL_python.dir\Release\vc141.pdb" /Gd /TP /wd4141 /wd4146 /wd4180 /w
  d4244 /wd4258 /wd4267 /wd4291 /wd4345 /wd4351 /wd4355 /wd4456 /wd4457 /wd4458 /wd4459 /wd4503 /wd4624 /wd4722 /wd4800 /wd4100 /wd4127 /wd4512 /wd4505 /wd4610 /wd4510 /wd4702 /wd4245 /wd4706 /wd431
  0 /wd4701 /wd4703 /wd4389 /wd4611 /wd4805 /wd4204 /wd4577 /wd4091 /wd4592 /wd4319 /wd4324 /wd4996 /errorReport:queue  /bigobj -w14062 /Zc:sizedDealloc- "C:\Softwares\DeepLearning\ELL-master\build\
  interfaces\python\ELLPYTHON_wrap.cxx"
  ELLPYTHON_wrap.cxx
C:\Softwares\DeepLearning\ELL-master\build\interfaces\python\ELLPYTHON_wrap.cxx(173): fatal error C1083: Cannot open include file: 'Python.h': No such file or directory [C:\Softwares\DeepLearning\EL
L-master\build\interfaces\python\_ELL_python.vcxproj]
Done Building Project "C:\Softwares\DeepLearning\ELL-master\build\interfaces\python\_ELL_python.vcxproj" (default targets) -- FAILED.

lovettchris, 2017-09-27:
We require Python 3.6, see [INSTALL-Windows.md](https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md) section under Python where we recommend you use Miniconda or Anaconda python environments. 

So for example, what I do is I use Anaconda, then I create a new environment called py3.6 using conda **create -n py36 anaconda python=3.6**, then I activate that using "**activate py36**" which gives me a console window that is now setup so that Visual Studio can find Python.h among other things.

When I run the rebuild.cmd script from there and everything should be good.

lovettchris, 2017-10-02:
closing due to inactivity, please reopen if you still have issues, thanks.

closed 2017-10-02 by lovettchris

amitmc19, 2017-10-03:
Apologies for the delay in my response. I have anaconda installed. To rule out installation problems, I uninstalled and reinstalled Anaconda.(conda version is 4.3.27)
I opened a terminal(win cmd prompt) using the Anaconda navigator and created a python 3.6 environment. I activated the python 3.6 environment. Post this I ran the script "cmake --build . --target _ELL_python --config Release" (as mentioned in the [link](https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md))

I still get the Python.h not found error. (Attached the build logs)
[pyELLBuildError.txt](https://github.com/Microsoft/ELL/files/1352767/pyELLBuildError.txt)

amitmc19, 2017-10-03:
I deleted the build folder, and followed the "Building ELL" section again. After this, I don't see the problem anymore. 
Thanks for the help. (Looks like the reinstall of anaconda helped)

lovettchris, 2017-10-03:
great!

SaimaSafdar, 2018-04-19:
Hello
I am getting lots of error while building slicer. I am new here. Please see the following:
I am using Visual studio 2017, git2.16.2-64-bit, cmake 3.11.0, qt 4.7.4, slicer 4.9.0, sil.1.9.7 and window 7

Please see the below:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C1083	Cannot open include file: 'Python.h': No such file or directory [C:\SLicer4Build\CTK-build\PythonQt-build\PythonQt.vcxproj] [C:\SLicer4Build\CTK-build\PythonQt.vcxproj]	CTK	c:\slicer4build\ctk-build\pythonqt\src\pythonqtpythoninclude.h	90	
Error	MSB6006	"cmd.exe" exited with code 1.	curl	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncher.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncherW.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncher.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	LNK1112	module machine type 'x86' conflicts with target machine type 'x64' [C:\SLicer4Build\CTKAppLauncherLib-build\CTKAppLauncherW.vcxproj]	CTKAppLauncherLib	C:\SLicer4Build\QtCored4.lib(QtCored4.dll)	1	
Error	MSB6006	"cmd.exe" exited with code 1.	python	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-appdirs	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-pyparsing	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-six	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-packaging	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-setuptools	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-smmap	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-gitdb	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-wheel	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-GitPython	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	python-nose	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-couchdb	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-chardet	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3.	python-pip	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	NUMPY	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	VTKv9	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 3. [C:\SLicer4Build\ITKv4-build\Wrapping\Modules\ITKCommon\ITKCommonSwig.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	C1083	Cannot open include file: 'patchlevel.h': No such file or directory [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyBase\ITKPyBasePython.vcxproj]	ITKv4	c:\slicer4build\itkv4\wrapping\generators\python\patchedpython27pyconfig.h	2	
Error	MSB6006	"cmd.exe" exited with code 1. [C:\SLicer4Build\ITKv4-build\Wrapping\Generators\Python\PyUtils\ITKPyUtilsCastXML.vcxproj]	ITKv4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets	171	
[... cut ...]
Error	LNK1104	cannot open file 'vtkRenderingOpenGL.lib' [C:\SLicer4Build\SlicerExecutionModel-build\ModuleDescriptionParser\ModuleDescriptionParser.vcxproj]	SlicerExecutionModel	C:\SLicer4Build\LINK	1	
Error	LNK1104	cannot open file 'vtkRenderingOpenGL.lib' [C:\SLicer4Build\SlicerExecutionModel-build\ModuleDescriptionParser\ModuleDescriptionParser.vcxproj]	SlicerExecutionModel	C:\SLicer4Build\LINK	1	
Error		VTK was not configured to use QT, you probably need to recompile it	Slicer	C:\SLicer4Build\CUSTOMBUILD	1