microsoft/ELL #98: cannot find wrap folder in \ELL-master\tools

amitmc19, 2017-10-03:
I followed the https://github.com/Microsoft/ELL/blob/master/INSTALL-Windows.md to setup ELL on my windows machine. Post this, I am trying to run the first [tutorial](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/)

In this tutorial, I am not able to proceed beyond step 3, because my ELL-master\tools folder doesn't have the wrap folder.

How do I get this folder?
Thanks in advance
Amit

lovettchris, 2017-10-03:
It lives here: [https://github.com/Microsoft/ELL/tree/master/tools/wrap](https://github.com/Microsoft/ELL/tree/master/tools/wrap)

amitmc19, 2017-10-03:
ok. If I copy this folder into my local setup, then I get a different error. 
"../utilities/pythonlibs\find_ell.py../utilities/pythonlibs\find_ell.py is not found"
Looks like lot of contents in the ELL changed after I cloned it.  

Should I use the latest clone and re start the windows setup again?

lovettchris, 2017-10-03:
Yeah, it is easier to run Wrap.py from where it is and not try and move it.  It uses some ".." paths to find other stuff in the repo that it needs.

closed 2017-10-30 by lovettchris