microsoft/ELL #99: Traning CNN in ELL

Ppinjari, 2017-10-09:
Does ELL supports training a CNN from scratch with help of labelled dataset? Does anyone have written any python script to train CNN in ELL?

clovett, 2017-10-09:
No, ELL is not a CNN training system, it is a compiler for CNN's (and other ML algorithms) that optimizes the execution of a pre-trained model on a given target platform with a focus on embedded platforms, like Raspberry Pi.  ELL currently supports importing trained models from [CNTK ](https://docs.microsoft.com/en-us/cognitive-toolkit/)and [darknet](https://pjreddie.com/darknet/).

Ppinjari, 2017-10-10:
If it doesn't support training of neural network, then what is the use case of trainers (SGD, SDCA, etc) that ELL is supporting. (Trainers are defined in libraries/trainers/). Will ELL support training of CNN in near future?

lovettchris, 2017-10-30:
The trainers in the libraries folder are only there because they are unique to ELL and do not exist elsewhere.  There is no plan for ELL to duplicate CNTK or tensorflow training.  ELL is focused on optimizing the inference side for running trained models on embedded hardware.

closed 2017-10-30 by lovettchris