microsoft/ELL #100: Can this still run using laptop webcam ..?

ack72tdp, 2017-10-19:
I have used this previously few months back .. At the time, I was able to compile and run using my windows laptop and object recognization using my laptop webcam..

Now lot of things changed in the instructions and site...

Does any one can this still run on Windows using my laptop webcam to recognize objects? 
If so can some one provide instructions on how to do the do that part ....

I compiled on windows laptop and downloaded CNTK model. But I am not sure how to run it..

(py36) C:\ELL\tools\importers\CNTK>python "cntk_import.py" model.cntk
Traceback (most recent call last):
  File "cntk_import.py", line 18, in <module>
    import cntk_to_ell
  File "C:\ELL\tools\importers\CNTK\cntk_to_ell.py", line 18, in <module>
    from cntk import load_model
ModuleNotFoundError: No module named 'cntk'

 I don't know if this is supposed to automatically launch the camera or not .... 

Any help is appreciated.

lovettchris, 2017-10-21:
Yes, you can still use your laptop. [This Tutorial ](https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-on-the-Raspberry-Pi/) explains how, especially step 3 and the tutorial on [Importing Models ](https://microsoft.github.io/ELL/tutorials/Importing-models/) will help you with the CNTK problem above (which looks like you are missing the CNTK Python module).

ack72tdp, 2017-10-23:
Thanks. It works great..

closed 2017-10-23 by ack72tdp