microsoft/ELL #101: Error on running tutorial.py

ack72tdp, 2017-10-19:
(py36) C:\ELL\build\tutorials\tutorial1\host\build>python tutorial.py
Traceback (most recent call last):
  File "C:\ELL\build\tutorials\tutorial1\host\build\model.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\anaconda35\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named '_model'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tutorial.py", line 25, in <module>
    import model
  File "C:\ELL\build\tutorials\tutorial1\host\build\model.py", line 17, in <module>
    _model = swig_import_helper()
  File "C:\ELL\build\tutorials\tutorial1\host\build\model.py", line 16, in swig_import_helper
    return importlib.import_module('_model')
  File "C:\anaconda35\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_model'

Any help is appreciated. All the steps compiled and executed until this point ..

lovettchris, 2017-10-21:
You may have missed a "cd .." in there somewhere.  The tutorial.py should be in host, not in build, and same for model.py.  You should see a _model.pyd file in build\Release folder, and model.py loads this, but it will not find it if model.py is in the wrong place.  So I would go back and repeat the tutorial instructions, and look for the "cd .." instructions.

ack72tdp, 2017-10-23:
Thank you .. That worked fine..

closed 2017-10-23 by ack72tdp