microsoft/ELL #102: call model run error??

ack72tdp, 2017-10-20:
(py36) C:\ELL\tutorials\tutorial1\host>python call_model.py
Model input shape: [224, 224, 3]
Model output shape: [1, 1, 1000]
Traceback (most recent call last):
  File "call_model.py", line 78, in <module>
    input_shape.rows)
  File "call_model.py", line 40, in prepare_image_for_model
    if image.shape[0] > image.shape[1]:  # Tall (more rows than cols)
AttributeError: 'NoneType' object has no attribute 'shape'

lisaong, 2017-10-20:
Hi, 

This error is not obvious, but it typically means that the image input is invalid.  If this is the script from the tutorial (running on the host, not on the Raspberry Pi), this error means that coffeemug.jpg is not found.  

Try downloading coffemug.jpg from the tutorial (there should be a link to it) and saving it to the location of your script.  

lisa

PS: OpenCV's cv2.imread doesn't actually fail if the file isn't found, but one enhancement that can be made to the script is to check (via Python) whether the file exists before feeding it to OpenCV.

ack72tdp, 2017-10-21:
Thank you. It was indeed missing the image file and after copying it worked fine.. Also call_model.py file came like this ..and it chokes .. because width was missing .. I changed that too...
 resized = cv2.resize(cropped, (height, )) --> resized = cv2.resize(cropped, (height, width))

(py36) C:\ELL\tutorials\tutorial1\host>python call_model.py
Model input shape: [224, 224, 3]
Model output shape: [1, 1, 1000]
Category index: 291
Confidence: 0.9998146891593933

Now the next problem is I want to use webcam to capture the image and send it the prediction model..
I copied tutorial.py file from the site in to the same host folder. and run it can't find helpers modules .. Any ideas on what I am missing with helpers module...

(py36) C:\\ELL\tutorials\tutorial1\host>python tutorial.py
Traceback (most recent call last):
  File "tutorial.py", line 28, in <module>
    import tutorial_helpers as helpers
ModuleNotFoundError: No module named 'tutorial_helpers'

All the steps compiled and run fine to this point ..

ack72tdp, 2017-10-21:
It's resolved .. This copies the file as pi3 ... so it is not recognizing the tutorial helper module ... It should be either pi3/tutorial_helpers.py or hosts/turorial_helpers.py 

[Windows] copy ..\..\..\docs\tutorials\shared\tutorial_helpers.py pi3

closed 2017-10-21 by ack72tdp