microsoft/ELL #103: Running CNN model in ELL

Tryljs, 2017-10-27:
Dear Team,
Does ELL support running CNN models from darknet or CNTK?

lovettchris, 2017-10-27:
Yes it does, see the tutorial on [importing models ](https://microsoft.github.io/ELL/tutorials/Importing-models/) for examples.

Tryljs, 2017-11-09:
Thank you!  
In the ELL, is the darknet model running  with a ELL frame or a darknet frame ?  And  what's the use of the codes in the neural file?

lovettchris, 2017-11-09:
If by "frame" you mean the input image format, the ELL model is expecting the same format RGB that darknet is expecting (not BGR like CNTK and opencv).  This is why our demoHelper.py makes the BGR to RGB conversion optional, so in the darknet case you request pass bgr=false.  This is mentioned in the tutorial at the bottom:

> Note: Using a converted Darknet model from pjreddie will require OpenCV input reordered from BGR to RGB. tutorial_helpers.py and tutorialHelpers.h provide helper functions for Python and C++, respectively, that take an argument to reorder the input image from BGR to RGB.

Looking at the tutorial_helpers prepare_image_for_model the argument is named reorder_to_rgb, so in the darknet case you set this to true.

But in [demoHelper.py](https://github.com/Microsoft/ELL/blob/master/tools/utilities/pythonlibs/demoHelper.py) it is called "bgr" and is an option to DemoHelper, and so in the darknet case you set that to false.

Tryljs, 2017-11-09:
If i want to use c++ to call the darknet model, does the ELL can  provide some function interfaces to implement it?

kernhanda, 2017-11-09:
Yes, if you look at https://microsoft.github.io/ELL/tutorials/Getting-started-with-image-classification-in-cpp/, you can see that ELL creates a function called `model_Predict`, where `model` can be replaced by a name of your choice.

closed 2017-11-09 by kernhanda